prusa3d/PrusaSlicer #333: Crash (basic_string::_S_construct null not valid) when generating gcode

liftoff, 2017-06-06:
### Version
Slic3r-1.35.1-prusa3d-linux64-full-201706021450.AppImage

### Operating system type + version
Ubuntu 17.04

### Behavior
Crashes when generating gcode...

    ./Slic3r-1.35.1-prusa3d-linux64-full-201706021450.AppImage

    (slic3r.pl:1493): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

    (slic3r.pl:1493): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
    Info about Centrifical_7_Curve.stl:
    size:              x=84.453 y=85.866 z=7.000
    number of facets:  24012
    number of shells:  1
    volume:            12792.941
    needed repair:     no
    terminate called after throwing an instance of 'std::logic_error'
    what():  basic_string::_S_construct null not valid
    Aborted

Tried changing various settings to see if it made a difference.  Nothing I've tried so far seems to work.

#### STL/Config (.ZIP) where problem occurs

https://www.thingiverse.com/thing:2307920/#files (Centrifical_7_Curve.stl)

My config along with the .stl in question are attached:
[Slic3r Crash Files.zip](https://github.com/prusa3d/Slic3r/files/1055607/Slic3r.Crash.Files.zip)

For reference, other files work just fine.  Just completed a whole bunch of prints 😄

bubnikv, 2017-06-06:
Interestingly, it works flawlessly with the Windows 10 64bit build.

liftoff, 2017-06-06:
Discovery: I tried changing from my "PETG" profile to "0.20MM NORMAL" and that doesn't crash Slic3r when generating gcode.  I'm going through the settings now to figure out what the difference is that causes the crash.

bubnikv, 2017-06-06:
Thanks. I tested on Ubuntu and I was able to slice with your settings without any issues.

There seems to be some issues likely with the newly added print extruder parameters. Maybe these issues are related?
https://github.com/prusa3d/Slic3r/issues/325

liftoff, 2017-06-06:
Aha!  It's the Autospeed settings that cause the crash.  If I set set anything other than 0 for "Max volumetric slope positive" or "Max volumetric slope negative" it crashes *but only for that model* (and the other .stl files for that Thingiverse thing, actually).  It doesn't cause crashes with any other models that I've tested.

bubnikv, 2017-06-06:
Great, thanks for the investigation. The volumetric extrusion smoothing is a highly experimental feature, which needs yet a lot of attention.

bubnikv, 2017-06-06:
I am closing this issue, it will be referenced from the other issue on the same topic:
"Automatic speed & volumetric extrusion slopes experimental features are broken"
https://github.com/prusa3d/Slic3r/issues/45

Thanks for investigation.

closed 2017-06-06 by bubnikv