prusa3d/PrusaSlicer #334: Preview: Extruder Color selector not showing

pcsentinel, 2017-06-06:
### Version
_1.35.1-prusa Win64

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows x64

### Behavior
when running program, app starts in full screen mode, the selection box for extruder color is not visible, if you resize the window, it then becomes visible and if you switch back to full screen remains visible
screen resolution is 1366 x 768

[image]
[image]

bubnikv, 2017-06-06:
I can reproduce it, thanks. It seems to be a pretty stupid bug in wxWidgets UI library.

pcsentinel, 2017-06-10:
Also getting with 1.35.2

Use of uninitialized value $color in pattern match (m//) at C:/Users/All/Downloads/Slic3r-1.35.2-prusa3d-win64-full-201706061718/Slic3r-1.35.2-prusa3d-win64-full-201706061718/lib/Slic3r/GUI/Plater/3DPreview.pm line 227.

Sorry if this is a double post, thought I had posted a separate report but I must have missed the save button somehow.

bubnikv, 2018-05-11:
This issue was a quirk of wxWidgets, it fixed itself once we extended the UI with the redo buttons. 

The fix will be in 1.40-alpha, which will be released soon. Closing.

closed 2018-05-11 by bubnikv