prusa3d/PrusaSlicer #336: "Arrange" not working

cerdeira, 2017-06-06:
Version
1.35.2 win64

Operating system type + version
Win 10 64 home 

 Behavior
After messing around with several parts on the plate, seleting auto Arrange  (3D view) does not do nothing.

bubnikv, 2017-06-06:
May it be a duplicate of https://github.com/prusa3d/Slic3r/issues/285 ?

cerdeira, 2017-06-07:
It  seems to be different.
I was moving parts around to get the best distribution on the plate to minimise printing time and  tried Arrange but parts do not move.

[image]

bubnikv, 2017-06-08:
This is really a different issue. The algorithm for automatic placement Slic3r uses is very limited: It measures the biggest object, then it draws a regular grid with cells as big as the biggest object, and then it places the objects into this regular grid. Because of the large object on the bed, the grid has big cells and there are not enough cells on the bed.

I will have to rewrite the placement algorithm one day.

cerdeira, 2017-06-09:
If it helps and following what I said on slic3r #4008, Auto arrange has a great potential.
It could be used to just laydown parts on plate, close toghether and aligned, or to move them around to minimise print time, or to group them by height to allow the user to aply mesch modifiers by height or layers, to group them by size or area, etc.

bubnikv, 2017-06-13:
I have ported the change made by @alexrj to do at least something if the objects do not fit the print bed.
That is not ideal, but it may be better than nothing for some.
https://github.com/prusa3d/Slic3r/commit/a4992602eef7e19721efebf145a3361af227978e

cerdeira, 2017-06-16:
Coming back to this subject, I noticed that other software do not do much either, they just align the objects by X and Y axes keeping them apart by 5mm or so.
A good improvement for Arrange could be align objects also by Z eight, avoiding a short object placed in between two tall ones.

fiveangle, 2017-07-17:
FYI, when you revisit redesigning the placement algo, you may want to review this project to see if it is something you want to invest the time to leverage:

https://github.com/Rhoban/Plater

I often use it myself to auto-arrange full plates and it is quite convenient.

bubnikv, 2017-07-17:
Thanks for heads up, Dave. It is C++, so we can hopefully use it straight
away.

nospoon, 2018-05-10:
Arrange doesn't seem to adhere to buildplate space, it just scatters the parts all over the place with huge gaps between them, so that some objects end up outside the buildplate.
[image]

bubnikv, 2018-05-14:
We have @tamasmeszaros working on a new placement algorithm. Tamas follows the papers referenced by http://svgnest.com/ , so we should get similar results in Slic3r.

bubnikv, 2018-07-17:
@tamasmeszaros implemented an advanced placement algorithm into Slic3r 1.41.0-alpha, which is due in a couple of days. Closing.

closed 2018-07-17 by bubnikv