prusa3d/PrusaSlicer #338: gcode generated by slic3r 1.35.x is no more waiting for set bed temperature is reached

rmoravcik, 2017-06-07:
### Version
Slic3r-1.35.2-prusa3d-linux64

### Operating system type + version
Ubuntu 17.04 64 bit

### Behavior
Starting from slic3r version 1.35.1, my printer stopped waiting for set temperature of bed is reached.

So, I've checked the generated gcode and all version before 1.35.1 was generating start gcode using M190, but the latest versions are using M140.

I don't know if this is an regression.

#### STL/Config (.ZIP) where problem occurs
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1058886/config.ini.zip)

bubnikv, 2017-06-13:
Reverted the "set_and_wait_temperatures" change  introduced in 1.35.1, as it was not in line with the latest @alexrj Slic3r and it caused confusion.

Please retest with the upcoming Slic3r 1.35.3. If that does not solve your issue, please reopen this issue.

closed 2017-06-13 by bubnikv

rmoravcik, 2017-06-13:
Thank you, just downloaded version 1.35.3 and it's working like it was before