prusa3d/PrusaSlicer #340: x-axis artifacts after slicing

aphex3k, 2017-06-08:
### Version 1.35.2-prusa3d

### Mac OS 10.12.5

### Behavior
* When slicing the preview window shows artifacts on the exterior shell and sure enough the gcode reflects that. I don't really know how to specifically describe that behavior, please see the screenshot! In the screenshot you can see the little lines on the lefthand side and that is interestingly the only corner/axis with such artifacts.

[image]

#### STL/Config (.ZIP) where problem occurs
[stl_config.zip](https://github.com/prusa3d/Slic3r/files/1062130/stl_config.zip)

aphex3k, 2017-06-08:
Unchecking the "Detect thin walls" setting solves the issue...

ahalekelly, 2017-06-08:
Can confirm I've also seen this issue.

bubnikv, 2017-06-08:
This is a similar topic:
https://github.com/prusa3d/Slic3r/issues/149

bubnikv, 2017-06-08:
and another: https://github.com/prusa3d/Slic3r/issues/227

> Can confirm I've also seen this issue. Seems to be that the wall is thin enough to qualify as a thin wall, but for some reason only at certain Z heights.

Exactly. 

This is a quite difficult problem to solve, also the thin extrusions are separate pieces, which do not bond to the outer perimeter well. We at Prusa Research have this feature disabled for our customers. To make the feature useful, two things would need to be solved:

1) Connect the thin wall to the outer extrusion, so the thin wall is extruded in a single draw with the outer perimeter

2) Fix your problem with the thin wall appearing / disappearing randomly.

aphex3k, 2017-06-08:
thanks for the comments. I'm not experienced enough to relate all these issues together. For now, I will deactivate the feature whenever I see something off in the preview...

clintonefisher, 2020-03-07:
I'm seeing this too. 
Here is the STL view:
[image]

Here is Detect Thin Wall turned OFF:
[image]

Here is Detect Thin Wall turned ON - Note the artifacts:
[image]

For repro testing and anaysis, the STL is located at:
https://www.thingiverse.com/download:7399937
(associated with this project:  https://www.thingiverse.com/thing:4048193/files)

PaparazziN, 2023-10-22:
Hi @aphex3k!
Have you tried using the 'new' Arachne Perimeter generator? I tried it with the file you attached and it comes out a lot better with the Arachne compared to the Classic generator with thin wall detection enabled. Based on the similarities with other issues and the age, it might be worth considering closing this issue.

closed 2026-09-29 by prusaslicer-bot[bot]