prusa3d/PrusaSlicer #341: Support on object sticks too well on bottom. Separate Contact Z Distance for top and bottom?

ahalekelly, 2017-06-08:
When printing a support material on top of the main object with a 0.2mm Contact Z Distance the top of the support separates easily, while the bottom sticks to the object more firmly. What could be the cause of this? Would it be possible to add a separate Contact Z Distance parameter for the base of support material that's on top of the main model?

bubnikv, 2017-06-08:
We have discussed a separate bottom Z distance internally at Prusa Research. You are right, this would likely improve the supports.

cerdeira, 2017-06-09:
Could it be possible to also consider the possibility to allow the user to modify support structures deciding where he wants or does not want support material?
This would be a huge improuvement on Slic3r.
It is the only thing I miss.

bubnikv, 2017-06-09:
> Could it be possible to also consider the possibility to allow the user to modify support structures deciding where he wants or does not want support material?

This has been requested many times and it has a high priority for me.

cerdeira, 2017-06-09:
That is good

cmfcmf, 2017-08-05:
This is the one biggest issue I am having with supports. The top breaks off very easily 👍, but the bottom sometimes almost is impossible to remove from the model.

liftbag, 2019-05-22:
I was about to open a feature request on the subject, when I found this topic still open.
I wonder if it is not appropriate to realize it in the next 2.0 releases.
For some filaments it is almost impossible to find a contact Z distance that allows to easily detach the low supports built on the top layer of the object, and at the same time have a good appearance of the bottom layer in contact with the top supports interface (because they are too far from each other).

bubnikv, 2019-05-22:
I agree, we shall do that soon.

szaghi, 2019-05-22:
I agree with @liftbag

liftbag, 2019-05-22:
Also, very useful would be to be able to separately establish the number and spacing of the interface layers for above and below the supports. It would often be convenient to have interface layers only above the supports.

bigsubgit, 2019-05-22:
Me too, I agree with @liftbag

bubnikv, 2019-05-22:
I agree with @liftbag @bigsubgit @szaghi

StevePC, 2019-11-14:
This is an absolute must.  In some prints, I can't get the interface material off at all, due to tight spaces/angles...

tychop, 2020-04-25:
*Bump* to keep this request alive.

RiriSe11, 2020-05-19:
I think it could already be helpful to turn off the interface layers for the bottom only. So there is no interface between Model and Support on the bottom side of the support structure

paulamath, 2020-10-19:
I agree with @liftbag @bigsubgit @szaghi @bubnikv

Sleslie706, 2020-11-19:
Yes I am also finding it difficult to remove bottom support interfaces when printed on print surfaces.

ricky-davis, 2021-01-14:
I either need some form of this feature, or tree supports for FDM. I've been trying to scrape the supports off for hours

spineytoad, 2021-02-05:
Same here, this was brought up in 2017...still not a feature.  Otherwise this PrusaSlicer is excellent!

bubnikv, 2021-03-16:
73b88e6ce0fa21a031490f148751114af73d573c

    Splitting FDM support gap to top / bottom, introducing
    support_material_bottom_contact_distance

[image]

closed 2021-03-23 by bubnikv

nickoe, 2022-01-05:
@bubnikv Has this change been removed again? I don't see the option in PrusaSlicer 2.3.3 -- even in the export settings.