prusa3d/PrusaSlicer #342: Explicit seam position and quality improvement

Gigga-Wc, 2017-06-09:
### Version
1.35.2/ All versions

### Operating system type + version
Windows 10 pro 64

### Behavior
There is no possibility to lay the seam in the inside of a horseshoe shape.
So i thought it would be great to implement a new feature, maybe in the objekt setting - modifier Task.
This would look like this:
[image]
[image]
in this screenshots, I added two cylinder which represent lines for a closest seamposition.

 

There also some defects in the seam at all, so i did some Research work.
 
* 1 Perimeterloops are not completely closed (No quality loss)
[image]
there is a gap of approximately 0.06mm.

 * 2 Move to startposition of external Perimeters should not be done in  rapidmove speeds 
   (medium to high quality loss ??known issue??)

 * 3 Coast feature (high quality improvemnt -known issue-)

 * 4 Extra length on Perimeter path after coast to blend all together 
    (low to medium Quality improvement -new feature-)
[image]

 * 5 Spiral way  Perimeters for smoothest toolpath  (medium quality improvement -new feature-)
[image]
I made this toolpath with my cad Software, and yes its kinky, because i was to lazy to add more than 
10 points and offsets / perimeter.  But if a Software can handle this it should spiral all the loops.
[image]
I dont know if its even possible on higher complex shapes, but a nice to have.
Its also not necessary at all. Mayby for a far future.

I add a zip with all gcodes and screenshots i have done.
Gcode is made for 0.4mm nozzle, PETG, 0.1mm layer and takes around 5min to print / test part

Thanks a lot for this nice Software and excuse me for bad english
Christian 
 
[Seam.zip](https://github.com/prusa3d/Slic3r/files/1062931/Seam.zip)

dartrax, 2018-02-14:
Hi Christian,
thank you for suggesting "Explicit seam position". We also need that feature to hide seam at the inside of some kind of housing. "Rear" is not working for this kind.

I wonder if someone started to implement this in the past, since I found a comment [in the code](https://github.com/prusa3d/Slic3r/blob/f38e0f2b4fd756c01111a47aac193a54d12d4739/xs/src/libslic3r/GCode.cpp#L1740):
`// Give a negative penalty for points close to the last point or the prefered seam location.`

So it may be easy to implement that into Gcode generation, but it also needs to be added to the UI.
Is anyone into that?

For the other feature suggestions I would recommend opening seperate issues for them.

dartrax, 2018-02-18:
This weekend I've dig inte the code and implemented the feature to set the prefered seam location:

[image]

[image]

As you can see in the pictures, a new modifier "lambda-seam" can be specified and seam will be aligned nearest to this modifier.
The rule is as follows:
If there is no modifier, "rear" will be used. If there is a modifier, it will control the preferred seam location for this and every layer above up to the next modifier.

This is my first contribution to Slic3r and also my first contribution in any open source project ever - so I'm still learning how this things work.
So, unfortunately, I was not able to commit/push the changes. I get the error message:

```
remote: Permission to prusa3d/Slic3r.git denied to dartrax.
fatal: unable to access 'https://github.com/prusa3d/Slic3r.git/': The requested URL returned error: 403
```

So I hope anybody can help me out ;-)

Gigga-Wc, 2018-02-18:
Hi dartrax,
nice job! I can't belive anyone cares about my request, since it's 8 months ago...
I'm not a developer either, so i can not help you.

I print a lot of threads, so I wonder if it's possible to lay the seam farest away from your lambda-seam point, this would save a lot time for cleanup.
Maybe you can check this out after your github problems are solved.

Thanks, Christian

Sebastianv650, 2018-02-18:
@dartrax do you tried pushing to this (prusa3d) repository? This will not work, as you don't have the rights to do this. You have to fork this repository to your own github account and do the changes in a new branch there. Than create a pull request to prusa slic3r.

I don't want to go into detail, because I'm bad in github on my own so I don't want to teach anybody wrong things. But thats the general process and I think you can find your way from there 👍

dartrax, 2018-02-18:
@Sebastianv650 thank you for that hint, that was exactly what I needed!

@Gigga-Wc This should be possible, but first let's see if the current implementation is working and if project maintainers agree with the way it's done.

Gigga-Wc, 2018-02-18:
Thank you dartrax, great job!
I saw your pullrequest so i call this done and close this issue.

Ps: can u upload a compiled version to you github site, so i can try out you work?

closed 2018-02-18 by Gigga-Wc

Gigga-Wc, 2018-02-18:
@dartrax i see, it is more complicated than i tought.
Anyway, thank's a lot

supermerill, 2018-02-19:
>  I can compile and run slicer on my pc with all the dependencies installed, but I could not figure out how to pack that into a working redistributable package.

Last time i fiddle with that:
 - copy a working release (the one you fork)
 -  replace the dll from lib\auto\Slic3r\XS with your own (search XS.dll).
 - replace all .pm you modified (it's in lib\slic3r i think)
 - test it

Add a .bat in the main repo to launch slic3r with perl (wperl.exe slic3r.pl) and delete the .exe

It's not the more robust way but it works for me, useful to give some test version to other people.

dartrax, 2018-02-19:
Actually I tried that yesterday and I got slicer to show for approximately 500ms before it crashed ;-)
I think I need to replace a lot more .pms since the main repo that I cloned already contains many more changes and features (which are great by the way) by others than in precompiled builds downloadable.
I will try that again tomorrow with some syncing tool to automatically find and replace all .pm files.
Thanks for the suggestion, so I know that I'm not on the wrong track.

supermerill, 2018-02-19:
If your change are not overwhelming : just cherry-pick / rebase your commits on the last version head.

for git-novice:
 - create a branch at the last release tag.
 - checkout to that new branch
 - cherry-pick (if using tortoisegit, right-clic on the commit in the  log window -> cherry-pick) your commits one after another. Resolve conflicts if needed.
 - done! you now just have to copy your changes.

dartrax, 2018-02-20:
Thank you!
I've managed to get a working package and currently I'm testing it at work. I'm not sure if I can upload it into my fork safely without committing accidentially next time I contribute something, so I'll take the short way of providing the dropbox link... Hope that's ok.
[Slic3r-1.39.1-beta-dartrax-win64-full-20180219.zip](https://www.dropbox.com/s/3z0nk6y9fvhqcbj/Slic3r-1.39.1-beta-dartrax-win64-full-20180219.zip?dl=0)
Make sure you've set seam positioning to "rear". The seam location can easily be checked in Preview Tab after Export of G-code. Note that the seam will not be positioned in an overhanging place.

dartrax, 2018-04-28:
I've uploaded an updated compiled version for Win 64bit on my release page:
https://github.com/dartrax/Slic3r/releases/tag/version_1.40.0-alpha