prusa3d/PrusaSlicer #344: Remove the 2D platter

bubnikv, 2017-06-09:
I propose to remove the 2D platter. I have never used it, neither my coworkers.

Would you guys please share your opinion?

Hugoz12, 2017-06-09:
I like to use it to arrange multiple parts (when arrange doesn't work how I want it).
I also use it on my Pi because the 3D is too slow for it when I have to rotate a bigger parts (smaller ones are ok even in 3D view). I select 2D view then add the part and just slice it without viewing in 3D to save time. (Sure I could use the terminal for it but I like to have a window.)

Maybe add a menu option to hide the 2D view (make it hidden by default). At the same place where you set auto enter etc.

Is there a certain reason to remove it? (Other than remove the tab for a cleaner UI, which could be achieved with the menu option.)

cerdeira, 2017-06-09:
I agree with removing it. It is a reminder of old days of slic3r, not up to today industry standards.
Besides we have the 3D view, the Preview, we even have the layer option.

I have never used 2D view.

bubnikv, 2017-06-12:
ok, I will leave it in for the Raspberry PI users for now. I have just reworked the part which calculates the 2D proxies to run in the main thread and to be triggered on demand, as the multi-threaded implementation was not correct and crashed.

closed 2017-06-12 by bubnikv