prusa3d/PrusaSlicer #345: Infill runs into very tight gaps with too much extrusion

bardiir, 2017-06-09:
### Version
1.34.1.24-prusa3d

### Operating system type + version
Mac OS X

### Behavior
When there is a very very small infill area then sometimes it seems like the infill doesn't calculate correctly that it should probably not be there.

When there is a small gap it does infill with a massively reduced extrusion multiplied as it seems
[image]

But when the gap is almost zero it bumps up extrusion very extremely, this is even worse on infills that follow the wall
[image]

When it finally reaches real zero space it's back to normal again
[image]

Hugoz12, 2017-06-10:
Does activating "Detect thin walls" solve this? I noticed something similar before and that fixed it for me. (But it may cause other problems.)

bardiir, 2017-06-10:
No detect thin walls is activated with this example but doesn't make any difference for the issue, only up to that point, the handling of the thinly extruded line differs.

bardiir, 2017-06-10:
Here is an AMF of that example:
[Test.amf.zip](https://github.com/prusa3d/Slic3r/files/1065822/Test.amf.zip)

bubnikv, 2017-06-12:
It looks like the "absolutely horrible" effect is caused by the Slic3r trying to fit a sparse infill into the thin gap.

bardiir, 2017-06-12:
@bubnikv - Sparse infill is when you combine infill layers right? Because with combine set to 2 it does this on every second layer. With combine set to 1 which would be sparse infill disabled it will do the same error but on every layer.

bardiir, 2017-06-13:
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1071304/config.ini.zip)

Here's my config that is causing this issue.

bardiir, 2017-06-13:
I just discovered that if I reduce the perimeter overlap down to 25% from its 55% default this effect vanished.

closed 2026-09-29 by prusaslicer-bot[bot]