prusa3d/PrusaSlicer #348: Horizontal holes cause perimeter issue

bardiir, 2017-06-10:
### Version
1.34.1.24-prusa3d

### Operating system type + version
Mac OS X 10.11.6 El Capitan

### Behavior
I'm seeing a slightly too wide wall near the top of the hole:
[image]

The model obviously does not have that:
[image]

#### STL/Config (.ZIP) where problem occurs
[Test.amf.zip](https://github.com/prusa3d/Slic3r/files/1065824/Test.amf.zip)

bardiir, 2017-06-10:
It does show up in the actual print too, but I'm not sure if it is related as on the print is does show up for more than just the two layers shown here.

bubnikv, 2017-06-13:
I just tested your STL and I don't see the effect.

Would you please provide your config.ini? Go to menu file->export config

Do you have the "resolution" value set at the end of the "Print Settings->Advanced" page to a nonzero value? That may lead to an artifact like this.

bardiir, 2017-06-13:
No resolution is set to 0, here's the config:
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1071306/config.ini.zip)

bubnikv, 2017-06-13:
> It does show up in the actual print too, but I'm not sure if it is related as on the print is does show up for more than just the two layers shown here. Could be a printing issue - in that case perfect representation of how it will actually print, but I doubt this is on purpose.

It is a bug in the visualization only. I hoped I fixed that already, but you found another occurrence of  it. It should not have any effect on the g-code.

bubnikv, 2017-06-14:
This is the other issue. https://github.com/prusa3d/Slic3r/issues/240

bubnikv, 2018-08-02:
It should be fixed with https://github.com/prusa3d/Slic3r/commit/3bedcf441390a12e10a3d6e1c9c94421c1dbb9e7
Closing.

closed 2018-08-02 by bubnikv