prusa3d/PrusaSlicer #349: disabeling "Ensure vertical shell thickness" leads to gaps

MartinMue, 2017-06-11:
### Version
1.34.1-prusa3d-win64

### Operating system type + version
Win 10 Home 64bit

### Behavior
When disabeling "Ensure vertical shell thickness" the top/bottom solid infill does not touch the shell.
TOP:
[image]
Bottom:
[image]

With feature enabled:
[image]

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
[3D.zip](https://github.com/prusa3d/Slic3r/files/1066360/3D.zip)

bubnikv, 2017-06-13:
Thanks. This is a known issue to me and it is caused by the "non-ensure vertical wall thickness" method producing too thin rings, where the infill pattern does not fit.

The "ensure vertical wall thickness" ensures that the ring is wide enough to be fillable, which then leads to error reports like 
"Ensure vertical shell thickness" is generating too much material #223 :-)

bubnikv, 2017-07-07:
This is really an issue with zero infill.
Another reports of this issue are following: 
https://github.com/prusa3d/Slic3r/issues/386
https://github.com/prusa3d/Slic3r/issues/253

bubnikv, 2017-07-07:
As this is a duplicate, I am closing this issue and keeping https://github.com/prusa3d/Slic3r/issues/253 open. Please feel free to post to https://github.com/prusa3d/Slic3r/issues/253 if needed.

closed 2017-07-07 by bubnikv