prusa3d/PrusaSlicer #350: Variable-speed

jornbyte, 2017-06-11:
The introduction of variable height is perfect.
I would still want the variable speed as very useful as with the variable height in percent.
If a bullet is printed from the bottom to the top of the bullet, 
it must be smaller to the center of the bullet to get a better result.

bubnikv, 2017-06-13:
> it must be smaller to the center of the bullet to get a better result.

I don't understand the request. Currently Slic3r decreases the print speed for small islands and for layers with short paths, and there is a request for Slic3r to slow down on overhangs. Does your request fit into these two scenarios?

jornbyte, 2017-06-13:
Yes, it's about slowing down the overhangs.
If this is still as adjustable as with the variable layer height it would be perfect.

bubnikv, 2017-06-13:
Then we can close this issue as it is a duplicate of https://github.com/prusa3d/Slic3r/issues/242 ?

bubnikv, 2017-10-03:
We have #242 for slowing down on overhangs. Closing this one.

closed 2017-10-03 by bubnikv