prusa3d/PrusaSlicer #351: Feature Request: Intelligent layer height setting

Quintox303, 2017-06-12:
The most common use of the variable layer height feature is to slice at finer layer heights where the top surface of the model is closer to being horizontal, and at coarser layer heights where the outer surface is closer to being just a vertical wall. 

Say that the top surface angle is 0 degrees when the surface is horizontal (flat) and 90 degrees when the surface is a vertical wall. Therefore the lower the angle, the lower we want the layer height to be. Slic3r could vary between the layer height limits defined in the settings intelligently such that it takes the lowest surface angle around the perimeter of a layer and defines the layer height based on that. Obviously the exception to this would be where the surface angle is zero in which case it would probably be best to use the same layer height as the layer below to maintain consistency.

I suspect that the surface angle detection wouldn't be too hard to integrate given that it could rely on the same system used to fine overhang angles for support.

This would be a great addition to what is already a very powerful tool. With the integration of intelligent layer heights, print quality could be drastically increased without any extra input from the user (like playing around with variable layer height settings).

bubnikv, 2017-06-12:
@platsch has already open a pull request for this feature,
https://github.com/prusa3d/Slic3r/pull/173
which I have to review. The general problem with this feature is, it usually works satisfactorily for simple objects only, and a perception based method likely needs to be implemented to work in general case.

Quintox303, 2017-06-12:
Oh someone already had the idea! Never mind then, hopefully that works out well

closed 2017-06-12 by Quintox303