prusa3d/PrusaSlicer #353: Help menu links do nothing

3axis, 2017-06-12:
### Version
_Version of Slic3r used goes here_
1.35.2
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 17.04 x64 appimage

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
Clicking on help menu items does nothing.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-06-13:
I just tested with Fedora 24 Workstation and it works. I have also tested on Windows 10 and OSX Maveric.  So this must be an Ubuntu specific issue. 

Slic3r calls the wxLaunchDefaultBrowser function to open a default web browser. Somehow this call fails on Ubuntu.

3axis, 2017-06-13:
ok thank you

bubnikv, 2021-03-12:
This is an old one reported for an obsolete Linux distro. Closing.

closed 2021-03-12 by bubnikv