prusa3d/PrusaSlicer #354: Automatic version check with message that there is a new version available

3axis, 2017-06-12:
### Version
_Version of Slic3r used goes here_
1.35.2

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
A automatic version check with message that there is a new version available would also be nice.

_Is this a new feature request?_
Yes 

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-03-14:
@vojtechkral is working on that.

bubnikv, 2018-08-02:
Implemented, working.

closed 2018-08-02 by bubnikv