prusa3d/PrusaSlicer #355: Avoid crossing perimeters

glottis512, 2017-06-13:
### Version
1.35.2-prusa3d-win64

### Operating system type + version
Win10 64bit

### Behavior
bad gcode generated with avoid crossing perimeters feature enabled. Result in screenshot. In gcode i can find this Z moves:
last good position: G1 Z34.150 F7800.000
and next Z move: G1 Z35.400 F7800.000

stl, bad gcode, config and screenshot attached

[example.zip](https://github.com/prusa3d/Slic3r/files/1071284/example.zip)

bubnikv, 2017-06-13:
I was not able to reproduce your issue. Is it reproducible at your side, if you import the config.ini, load the STL, slice and export the G-code?

It is quite strange, that a couple of layers are missing at your screenshot. I did not experience these missing layers.

glottis512, 2017-06-13:
ouch ... after i was importing config.ini is this error fixed! 

a jeste jedne veci jsme si vsiml a to uz ted nemam silu popisovat anglicky ani to ted overit, jeslti to je taky opravene. mezo asi 15 tou a 40 vrstvou tohoto objektu to zacne docela divocit a vyplnuje to zuby tim zpusobem, ze to docela rychle jezdi z vyplne zubu do stredu a zpatky. tim jak to pri tomhl epohybu nedela retract tak to nechava stopu a kresli to takovou hvezdu na infill. Bylo by to docela jedno ale uprostred se zacne hromadit material a zamatla to trysku. Zitra ve volnem case to dam tisknout znova jak to vypada ted. nebo existuje nejaky software co ukaze pohyb hotendu?

M

rlesac, 2018-11-26:
Crossing parameters works opposite for me and I get crazy unnecessary toolhead travels.

Avoid crossing perimeters disabled
[image]

Avoid crossing perimeters enabled
[image]

beegmouse, 2020-04-08:
I have a part this is causing issues with, this seems a very old issue, surprised it still persists.
For me it is particularly an issue for the internal perimeters.
Expected behavior (Simplify 3D)
[image]
Actual Behavior
[image]
The result is.
[image]

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

[image]

closed 2020-12-11 by bubnikv