prusa3d/PrusaSlicer #356: Problems with "support from the build plate only" function.

philggg, 2017-06-14:
### Version
_V1.34.1-prusa3d-win64_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 8.1

### Behavior
* When using "support from the build plate only there are errors in the layout:
1.  Not all areas are suppored
2. Some features are put ouside of the skirt.
* _Steps needed to reproduce the problem_
  * See attached stl files and resulting GCODE. (
* _Expected Results_
* notice the errors in support AND how some of the design is outside of the skirt
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_  no
[New Compressed (zipped) Folder.zip](https://github.com/prusa3d/Slic3r/files/1073179/New.Compressed.zipped.Folder.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-06-14:
I see the supports being generated. What exactly is the problem? 

[image]

If you would like the issue to be fixed, then please distill an as simple as possible test case. It is very hard to understand, on which of these 14 objects you think the support is not generated correctly.

philggg, 2017-06-14:
Sorry I made the example too complicated.  Here is a simplified version with one object.
The "arms" of the object should have support under them.  Compare the pictures from Cura with the picture from Slicr.
[image]
[image]
[New Compressed (zipped) Folder.zip](https://github.com/prusa3d/Slic3r/files/1074906/New.Compressed.zipped.Folder.zip)

bubnikv, 2017-07-07:
I believe the issue is, that the gap between the print bed and your object is too thin to fit a support and a prescribed gap between the object and the support interface. There is a difference between Cura and Slic3r in how they handle supported overhangs. Slic3r extrudes the horizontal surfaces over supports with bridging flow, making the object somehow thicker than what the source geometry dictates. Cura on the other hand extrudes the bottom horizontal layer over supports with normal flow, making the first layer over supports sparse.

bubnikv, 2021-03-18:
[image]

To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

With no "thick bridges", the supports fit below the object bottom.

closed 2021-03-18 by bubnikv