prusa3d/PrusaSlicer #357: Path planning inefficient when "plus" button is used to multiply objects

bubnikv, 2017-06-14:
When inserting multiple objects one by one, Slic3r optimizes the print path by sorting the objects by a greedy algorithm to a suboptimal "traveling salesman" path, but it is better than nothing.

When objects are duplicated with the "plus" button, even this simplistics optimization algorithm is not applied and the duplicate objects are picked randomly.

see
http://shop.prusa3d.com/forum/print-tips-slic3r-settings-kisslicer-model-repair--f12/printing-multiple-parts-inefficient-t4739.html
for further details.

ioquatix, 2017-11-07:
If the objects are different sizes, the biggest size is used when placing the objects. It makes the placement very inefficient. Manually placing objects works, but every time I add more objects, the entire layout is lost/reset.

bubnikv, 2020-02-19:
Fixed in PrusaSlicer 2.2.0-alpha4. Closing.

closed 2020-02-19 by bubnikv