prusa3d/PrusaSlicer #358: Layer Height Problems

cerdeira, 2017-06-15:
Version
1.35.3 prusa win64

 Operating system type + version
Win10 64 home

 Behavior
When you 1st slice an object after launching s3r prusa, layer heights are respected, but if you choose new layer height values some times it gets stuck with 0.1mm layer height. First layer is always ok.

Is this a new feature request?_
No

cerdeira, 2017-06-15:
Version 1.35.4 still have the same problem.

bubnikv, 2017-06-15:
Sorry, but I don't understand. Would you please be more specific?

bubnikv, 2017-06-15:
I see it now, thanks.

bubnikv, 2017-06-15:
This was an embarrassing one line bug.
Fixed with https://github.com/prusa3d/Slic3r/commit/cb43f19a90c05b5b8830b5594bc8e2f933d48dfc
A new build is on its way. Hopefully I may claim the 1.35.5 to be a final relese.

closed 2017-06-15 by bubnikv

cerdeira, 2017-06-15:
It is doing well no, thanks.

cerdeira, 2017-06-15:
It is doing well now, thanks.

Sorry about this mistake.

2017-06-15 21:39 GMT+01:00 Jose Cerdeira <guerra.cerdeira@gmail.com>: