prusa3d/PrusaSlicer #359: Too much support generated for "buildplate only" supports

jangrewe, 2017-06-15:
### Version
1.35.4 / x64

### Operating system type + version
Windows 10

### Behavior
* When generating "form buildplate only" supports for an object that has holes, supports will be generate for those holes - including support for the outside structure, although that wouldn't need support.

* __Expected:__ No supports are generated for the outside slope or the holes
* __Actual:__ Supports are being generated for the holes, and the outside slope below the holes (but not the complete slope)

[image]

#### STL/Config (.ZIP) where problem occurs
STL: https://www.thingiverse.com/thing:1696715/#files -> vive-rifle-front-controller.stl
Config: Stock from https://github.com/prusa3d/Slic3r-settings

bubnikv, 2017-07-07:
Fixed with https://github.com/prusa3d/Slic3r/commit/7c1350d007f3054905d3a8271e548ff3b257afea
The fix will be part of the Slic3r 1.36.x builds.
If you see this issue to reoccur, feel free to reopen this issue and add new findings.

freiser77, 2017-07-23:
It seems that the bug is not fixed in the 1.36.0 release. The arms of the puppet should not be supported as their slope angle is clearly beyond 30 degrees which is configured.
[Archiv.zip](https://github.com/prusa3d/Slic3r/files/1167936/Archiv.zip)
[image]
[image]

bubnikv, 2017-07-24:
It looks better if you set the "resolution" to zero.

[image]

[image]

bubnikv, 2017-07-24:
The explanation is, reducing the resolution of the slicing contours leads to spikes on the surface, and these spikes are rightfully recognized as overhangs, therefore they get supports. It is enough if there are tiny overhangs, and Slic3r will support them with a pillar.

jangrewe, 2017-08-12:
I still have this problem with 1.36.2 and Resolution also set to 0.

This STL is currently giving me a hard time: https://www.thingiverse.com/thing:2457996

bubnikv, 2017-10-04:
Another occurrence:
https://shop.prusa3d.com/forum/print-tips-slic3r-settings-kisslicer-model-repair--f12/slic3r-adding-unnecessary-supports-t5818.html