prusa3d/PrusaSlicer #360: Overhang threshold not utilized after 1.34.1

tcm0116, 2017-06-16:
### Version
1.34.1.24-prusa3d and later

### Operating system type + version
OS X 10.12.3

### Behavior
On all versions from 1.34.1.24-prusa3d to the current 1.35.5-prusa3d, the Overhang threshold support parameter is not utilized. Here are some screenshots showing some tests using https://www.thingiverse.com/thing:265529. As you can see, the newer versions of Slic3r pretty much only support overhangs of 10 degrees from horizontal or less, which is a significant problem.

1.34.1 - Overhang threshold = 0

[image]

1.34.1 - Overhang threshold = 60

[image]

1.34.1 - Overhang threshold = 10

[image]

1.35.5 - Overhang threshold = 0

[image]

1.35.5 - Overhang threshold = 60

[image]

1.35.5 - Overhang threshold = 10

[image]

#### STL/Config (.ZIP) where problem occurs

[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1079648/config.ini.zip)

bubnikv, 2017-06-16:
I believe this issue describes basically the same effect you are experiencing: https://github.com/prusa3d/Slic3r/issues/273

The reason why you are now seeing a difference since 1.34.1.24 is, you set the support_material_xy_spacing
to one meaning a gap of 100% of the perimeter line. This parameter was erroneously ignored before 1.34.1.24 for the gap of the contact layer and a fixed hard coded value of 50% was used instead. Now Slic3r uses this preset value for the contact layer as well, making the part of the overhang to be supported disappearing due to the trimming by the lower layers offsetted by the XY offset.

This is exactly the reason why the support_material_enforce_layers parameter has little effect for overhangs with steep angles, where the overhangs are smaller than 50% of the perimeter width.

I will look into this issue, but most likely not for this release.

tcm0116, 2017-06-16:
Thanks for the explanation. I did some more experimenting, and it does indeed appear that decreasing the XY separation allows for the support material to be generated for overhangs with higher angles. Hopefully you can find a resolution soon as having to decrease the XY separation can lead to the support material getting stuck to the part in areas that don't need to be supported.

bubnikv, 2017-07-07:
Fixed with https://github.com/prusa3d/Slic3r/commit/abcd746774cbb6a392b275e331621dbb6b459ded

closed 2017-07-07 by bubnikv