prusa3d/PrusaSlicer #361: Infill error with variable layer height

tcm0116, 2017-06-16:
### Version
1.35.5-prusa3d

### Operating system type + version
OS X 10.12.3

### Behavior
When using the variable layer height feature (via the Layer Editing), the infill in the areas with reduced layer height severely overlaps the perimeters, as if the infill is just being scaled and not re-computed for the layer height.

In this screenshot, I have the layer height set to 0.3, but I've used the Layer Editor to lower the layer height in this region to the minimum height. As you can see, the infill is severely overlapping the perimeters, which will likely result in a failed print in that area.

[image]

In this screenshot, I've reduced the layer height of the entire model to 0.06 and reset the Layer Editor. The infill is now correctly calculated.

[image]

#### STL/Config (.ZIP) where problem occurs

[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1081056/config.ini.zip)

Hugoz12, 2017-06-16:
Could you upload the stl and tell us at which height this happened?

tcm0116, 2017-06-16:
I can replicate this issue with just a standard cube. Use the Layer Editor to increase the detail to the maximum in the middle of the cube and then slice it. You'll see the same effect in that area when you view the Preview.

I did find that changing the Infill/Perimeters overlap from a percentage to a fixed value help out immensely.

Hugoz12, 2017-06-16:
Can confirm this. With a default extrusion width the problem seems to be gone for me.

Important: Remove your config and reupload it without your octoprint adress and give your server a password.

tcm0116, 2017-06-16:
> Important: Remove your config and reupload it without your octoprint adress and give your server a password.

Yikes! Thanks for the tip.

bubnikv, 2017-06-19:
The problem lies in the "sane" defaults Slic3r picks for the extrusion widths, if you leave the following values to zeros:

[image]

Even if you do not have the Prusa i3 MK2 printer, we recommend to base your settings on ours:
https://github.com/prusa3d/Slic3r-settings

bubnikv, 2017-07-19:
I have changed the default extrusion width to a 1x to 1.125x of the nozzle diameter.
https://github.com/prusa3d/Slic3r/commit/ac672d9dc592dc94ea1efdfeecdb75efc0a19233
This will give you much more sensible default widths.

There will still be the same issues you experienced, if the infill / perimeter overlap is given in percentage and the width of the infill extrusion varies from the width of the perimeter extrusion wildly.

bubnikv, 2018-05-18:
Is it still an issue?

When the overlap is defined with a percentage, it is calculated from the average of the infill extrusion width and the perimeter extrusion width (the perimeter with which the infill overlaps). If the infill extrusion width is much higher than the perimeter extrusion width, and the overlap in percents is high, than it may happen, that you will get the effects described at the beginning of this issue. I may change the rule to take the minimum of the two extrusion widths for the calculation of the overlap. Should I?

closed 2026-09-29 by prusaslicer-bot[bot]