prusa3d/PrusaSlicer #363: Repetier host has touble with gui-mode (v 1.35.x

E000001, 2017-06-17:
### Version
1.35.4 and 1.35.4

### Operating system type + version
Windows 10

### Behavior
After I updated my old version of Prusa Slic3r (previously 1.34.1) Repetier Host has trouble to open a windows with slic3r gui.

There are several lines from log. It seems, that there is not any option like gui-mode in slic3r. But When I try version 1.35.1 it is working.
  
18:59:10.354 : <Slic3r> Unknown option: gui-mode
18:59:10.355 : <Slic3r> Slic3r 1.35.5-prusa3d-win64 is a STL-to-GCODE translator for RepRap 3D printers
18:59:10.355 : <Slic3r> written by Alessandro Ranellucci <aar@cpan.org> - http://slic3r.org/
18:59:10.355 : <Slic3r> Usage: slic3r.pl [ OPTIONS ] [ file.stl ] [ file2.stl ] ...
18:59:10.355 : <Slic3r>     --help              Output this usage screen and exit
18:59:10.355 : <Slic3r>     --version           Output the version of Slic3r and exit
18:59:10.355 : <Slic3r>     --save <file>       Save configuration to the specified file
18:59:10.355 : <Slic3r>     --load <file>       Load configuration from the specified file. It can be used 
18:59:10.355 : <Slic3r>                         more than once to load options from multiple files.
18:59:10.355 : <Slic3r>     -o, --output <file> File to output gcode to (by default, the file will be saved
18:59:10.355 : <Slic3r>                         into the same directory as the input file using the
18:59:10.355 : <Slic3r>                         --output-filename-format to generate the filename.) If a
18:59:10.356 : <Slic3r>                         directory is specified for this option, the output will
18:59:10.356 : <Slic3r>                         be saved under that directory, and the filename will be
18:59:10.356 : <Slic3r>                         generated by --output-filename-format.
18:59:10.356 : <Slic3r>   
18:59:10.356 : <Slic3r>   Non-slicing actions (no G-code will be generated):
18:59:10.356 : <Slic3r>     --repair            Repair given STL files and save them as <name>_fixed.obj
18:59:10.356 : <Slic3r>     --cut <z>           Cut given input files at given Z (relative) and export
18:59:10.356 : <Slic3r>                         them as <name>_upper.stl and <name>_lower.stl
18:59:10.356 : <Slic3r>     --split             Split the shells contained in given STL file into several STL files
18:59:10.356 : <Slic3r>     --info              Output information about the supplied file(s) and exit
18:59:10.356 : <Slic3r>     
18:59:10.356 : <Slic3r>     -j, --threads <num> Number of threads to use (1+, default: 4)
18:59:10.356 : <Slic3r>   GUI options:
18:59:10.356 : <Slic3r>     --gui               Forces the GUI launch instead of command line slicing (if you
18:59:10.356 : <Slic3r>                         supply a model file, it will be loaded into the plater)
18:59:10.356 : <Slic3r>     --no-plater         Disable the plater tab
18:59:10.356 : <Slic3r>     --autosave <file>   Automatically export current configuration to the specified file
18:59:10.356 : <Slic3r>   Output options:
18:59:10.356 : <Slic3r>     --output-filename-format
18:59:10.356 : <Slic3r>                         Output file name format; all config options enclosed in brackets
18:59:10.356 : <Slic3r>                         will be replaced by their values, as well as [input_filename_base]
18:59:10.356 : <Slic3r>                         and [input_filename] (default: [input_filename_base].gcode)
18:59:10.356 : <Slic3r>     --post-process      Generated G-code will be processed with the supplied script;
18:59:10.356 : <Slic3r>                         call this more than once to process through multiple scripts.
18:59:10.356 : <Slic3r>     --export-svg        Export a SVG file containing slices instead of G-code.
18:59:10.356 : <Slic3r>     -m, --merge         If multiple files are supplied, they will be composed into a single 
18:59:10.356 : <Slic3r>                         print rather than processed individually.
18:59:10.356 : <Slic3r>   
18:59:10.356 : <Slic3r>   Printer options:
18:59:10.356 : <Slic3r>     --nozzle-diameter   Diameter of nozzle in mm (default: 0.5)
18:59:10.356 : <Slic3r>     --print-center      Coordinates in mm of the point to center the print around
18:59:10.356 : <Slic3r>                         (default: 100,100)
18:59:10.356 : <Slic3r>     --z-offset          Additional height in mm to add to vertical coordinates
18:59:10.356 : <Slic3r>                         (+/-, default: 0)
18:59:10.356 : <Slic3r>     --gcode-flavor      The type of G-code to generate (reprap/teacup/repetier/makerware/sailfish/mach3/machinekit/smoothie/no-extrusion,
18:59:10.356 : <Slic3r>                         default: reprap)
18:59:10.356 : <Slic3r>     --use-relative-e-distances Enable this to get relative E values (default: no)
18:59:10.356 : <Slic3r>     --use-firmware-retraction  Enable firmware-controlled retraction using G10/G11 (default: no)
18:59:10.356 : <Slic3r>     --use-volumetric-e  Express E in cubic millimeters and prepend M200 (default: no)
18:59:10.356 : <Slic3r>     --set-and-wait-temperatures Use M190 instead of M140 for temperature changes past the first (default: no)
18:59:10.356 : <Slic3r>     --gcode-comments    Make G-code verbose by adding comments (default: no)
18:59:10.356 : <Slic3r>     
18:59:10.356 : <Slic3r>   Filament options:
18:59:10.356 : <Slic3r>     --filament-diameter Diameter in mm of your raw filament (default: 3)
18:59:10.356 : <Slic3r>     --extrusion-multiplier
18:59:10.356 : <Slic3r>                         Change this to alter the amount of plastic extruded. There should be
18:59:10.356 : <Slic3r>                         very little need to change this value, which is only useful to 
[... 217 lines cut ...]
18:59:10.359 : <Slic3r>     --support-material-extrusion-width
18:59:10.359 : <Slic3r>                         Set a different extrusion width for support material
18:59:10.359 : <Slic3r>     --infill-overlap    Overlap between infill and perimeters (default: 55%)
18:59:10.359 : <Slic3r>     --bridge-flow-ratio Multiplier for extrusion when bridging (> 0, default: 1)
18:59:10.359 : <Slic3r>   
18:59:10.359 : <Slic3r>    Multiple extruder options:
18:59:10.359 : <Slic3r>     --extruder-offset   Offset of each extruder, if firmware doesn't handle the displacement
18:59:10.359 : <Slic3r>                         (can be specified multiple times, default: 0x0)
18:59:10.359 : <Slic3r>     --perimeter-extruder
18:59:10.359 : <Slic3r>                         Extruder to use for perimeters and brim (1+, default: 1)
18:59:10.359 : <Slic3r>     --infill-extruder   Extruder to use for infill (1+, default: 1)
18:59:10.359 : <Slic3r>     --solid-infill-extruder   Extruder to use for solid infill (1+, default: 1)
18:59:10.359 : <Slic3r>     --support-material-extruder
18:59:10.359 : <Slic3r>                         Extruder to use for support material, raft and skirt 
18:59:10.359 : <Slic3r>                         (1+, 0 to use the current extruder to minimize tool changes, default: 1)
18:59:10.359 : <Slic3r>     --support-material-interface-extruder
18:59:10.359 : <Slic3r>                         Extruder to use for support material interface 
18:59:10.359 : <Slic3r>                         (1+, 0 to use the current extruder to minimize tool changes, default: 1)
18:59:10.359 : <Slic3r>     --ooze-prevention   Drop temperature and park extruders outside a full skirt for automatic wiping
18:59:10.359 : <Slic3r>                         (default: no)
18:59:10.359 : <Slic3r>     --standby-temperature-delta
18:59:10.359 : <Slic3r>                         Temperature difference to be applied when an extruder is not active and
18:59:10.359 : <Slic3r>                         --ooze-prevention is enabled (default: -5)
18:59:10.359 : <Slic3r>     
18:59:10.439 : <Slic3r> Slic3r installed in a loclized path. Using an 8.3 path: "C:\PROGRA~1\REPETI~1\Slic3rPE\"

pax83, 2017-06-17:
The same problem, starting with Slic3r 1.35.4

bubnikv, 2017-06-19:
I have removed the gui-mode switch from Slic3r, and I have removed the "Simple" gui mode for good. This mode was too simple to be useful, therefore nobody used it.

I should have likely made this option to be ignored by the Slic3r.

bubnikv, 2017-06-19:
The option --gui-mode has been removed with the following option:
https://github.com/prusa3d/Slic3r/commit/82bde51c2752656aed517d89356272de277408b4

I have added this option for compatibility reasons:
https://github.com/prusa3d/Slic3r/commit/046d5a9cb77c103c7d323bed023ddc8412856ebc

closed 2017-06-19 by bubnikv