prusa3d/PrusaSlicer #364: Cut operation flips part

tcm0116, 2017-06-18:
### Version
1.35.5-prusa3d

### Operating system type + version
OS X 10.12.3

### Behavior
When using the cut tool to remove the top portion of a part, the part is flipped over. Steps to reproduce:

1. Add a the XYZ cube (https://www.thingiverse.com/thing:1278865) to the platter
2. Rotate the part 180 degrees around the Y axis such that the "Z" is on the bottom pf the part against the platter
3. With the object selected, select the Cut tool
4. Set Z to 10, uncheck Keep Upper part, and select Perform cut
5. The "Z" that was on the bottom of the part will now be on the top

#### STL/Config (.ZIP) where problem occurs

N/A

Hugoz12, 2017-06-18:
Did you uncheck "Lower part upwards"?

tcm0116, 2017-06-18:
> Did you uncheck "Lower part upwards"?

No I did not. I didn't realize what that option performed. It seems odd to me that it would be enabled by default.

However, there is still an issue in that "Lower part upwards" is not reflected in the preview when "Show preview" is selected.

bubnikv, 2017-06-19:
I believe it works as intended. When you cut the object, you likely want to
print it with the flat part on the bed.

bubnikv, 2017-06-26:
I believe it works as intended, closing.

closed 2017-06-26 by bubnikv

tcm0116, 2017-06-26:
@bubnikv should I make a separate issue for the"Lower part upwards" option not being reflected in the preview?

bubnikv, 2017-06-26:
> should I make a separate issue for the"Lower part upwards" option not being reflected in the preview?

Thomas, I don't think such a request will ever be implemented as there are much more pressing issues to implement or fix.