prusa3d/PrusaSlicer #366: Feature request for standby temperature

pcsentinel, 2017-06-18:
### Version
1.35.5

Feature request:
Can we have a standby temperature for the filament that can be used as a variable in the Custom GCode, I am using G10 and have to manually adjust at the moment, rather than assigning to filament profiles.

bubnikv, 2017-06-19:
Looking at http://reprap.org/wiki/G-code
7.5 G10: Tool Offset
7.6 G10: Retract
7.7 G11: Unretract

Would the "ooze prevention" not serve the same purpose?

[image]

pcsentinel, 2017-06-19:
Unfortunately not as it's filament dependant, so you maybe using a pla with active 190 and standby 170 and a support material that has active 215 and standby of 170. I generally use a standby temp which is the maximum temp where there is no nozzle ooze.

Also using Ozzie prevention automatically adds the shield which I don't need all the time.

pcsentinel, 2017-06-20:
I think I see your confusion, I am referring to the G10 that sets standby and active temps as per the S and R parameters

bubnikv, 2017-07-07:
Could you store your standby parameters in the custom filament specific start G-code then?
This is a new feature in Slic3r PE 1.35.5.

bubnikv, 2017-11-28:
No response, closing.

bubnikv, 2017-11-28:
Also there is a new conditional G-code syntax with math expressions, which will likely allow one to implement the required behavior.

closed 2017-11-28 by bubnikv

apphead, 2018-06-06:
I agree with pcsentinel. While ooze prevention combined with math expressions presents a feasible way to solve the issue, it is a) not very user-friendly b) actually high-jacks the ooze delta temp parameter for something else and c) the per filament standbytemp would come in handy during meshbed leveling to prevent oozing (as discussed [here](https://shop.prusa3d.com/forum/others-archive--f82/dealing-with-ooze-during-mesh-bed-leveling--t5582.html)). Not to mention that dualx setups would immensly profit from such an easy to set approach during toolchange / parking.
And as pcsentinel already wrote, ooze shields are not always the right solution (e.g. for IDEX with one head properly parked on the side of the bed, ooze shield and towers become obsolete). 
Hopefully these are good reasons to implement the additional parameter (which should be fairly easy to do as it has zero impact on the code except the scripts of course....)
PS: Love the new features of  1.40 beta release! Keep it up!

apphead, 2018-06-06:
Tried this toolchange script, but the parser did not like it at all:

`M104 T[previous_extruder] S{[temperature_[previous_extruder]]+[standby_temperature_delta]} ;lower current extruder temperature`
`M104 T[current_extruder] S[temperature_[current_extruder]] ;reset printing temperature`

apphead, 2018-06-07:
> M104 T[previous_extruder]
S{[temperature[previous_extruder]]+[standby_temperature_delta]}
;lower current extruder temperature

parser is not happy because of calculation with placeholders. "standalone" calculation works fine with curly braces. But nevertheless, this could be done easily and with the sideeffects outlined above by adding the standby temp to the filament properties and expose it as a placeholder....
There a reason not to implement it? Ever since I use 170°C during bedleving I had zero issues with oozing. That should be reason enough for Prusa to implement it, no?

bubnikv, 2018-06-07:
> parser is not happy because of calculation with placeholders.

Please post an error log.

> "standalone" calculation works fine with curly braces. 

I don't understand what you mean with "standalone"

The best thing you can do is to provide us with your complete configuration, so we can test it.

apphead, 2018-06-08:
Just put the lines of code below in Printer Settings > Custom G-codes > Tool change G-code and export gcode

- As for the standalone calculation:
M104 T[previous_extruder] S{200+22} //works
- As for the calculation with placeholder as per your suggestion
M104 T[previous_extruder] S{[temperature[previous_extruder]]+[standby_temperature_delta]} //fails

BUT you are missing my point! As the title says, adding standby temp to the filament properties and expose it as a placeholder is what this feature request is about.

jjaeggli, 2020-02-05:
A standby temperature as a per-material property which can be used in custom gcode can be thought of in a number of different ways. You could define it as  "ooze threshold temp" and can either be thought of as the maximum temperature at which no significant material emerges from the nozzle spontaneously. As it relates to the material and not to the print setting this could be applied to such things as:

*  stand-by temperature for multiple extruders
*  probe temperature for conductive calibration
*  stand-by temperature for pauses (in situations where priming wouldn't be appropriate)

If used properly, other temperatures could be easily derived from this with offsets, such as soften / wipe temps (unlike printing temperature). As an attempt at this I produced some coefficients using regression which approximates this for some materials, but could be putting additional strain on the extruder and produces subpar calibration results:

{temperature[0] * 0.848 + bed_temperature[0] * 0.135 - 40.8}

This produces temperatures which are both too low (MH PLA Pro) and too high (PETG). See: [sheet](https://docs.google.com/spreadsheets/d/1qBnuBxTFkwxzAvIyzxTNe7HdbsV6XBcxClWo2N2Htvg/edit?usp=sharing)

This would be useful for defining probe / wipe temperatures in the LulzBot profiles here:

https://github.com/jjaeggli/prusaslicer-lulzbot-taz

bubnikv, 2020-02-06:
Do I understand your request correctly? I just want us to add a
"standby_temperature" field to the filament presets?

Do you want slicer to apply this value automatically? Or do you want to
handle it in the custom scripts?
I am aware of only one place where slicer may apply the value
automatically, that is by the ooze prevention.

By the way, nice profiles. Do you want them to be added to PrusaSlicer
installation wizard?
@rtyr is responsible for PrusaSlicer's profiles.

jjaeggli, 2020-02-07:
Correct, I would like to add a standby_temperature, and for my purposes I would add this to custom scripts or start and end gcode. This would be done manually. If you think this would be useful, either for MMU or dual extrusion setups, it could be used by the slicer itself, but I'm not making this proposal. I think that unless it has a direct impact on slicing, it should probably not show up in the UI as it might confuse users who think they're changing something which has an effect on printing. If this is something which is used by the slicer, we should come up with a consistent definition of what this means so that we don't need separate materials defined for printers which use this value in different ways.

As far as how this fits into your backlog and prioritization, I could send you a pull request if this is not something you think you will get to right away.

Thanks for taking a look at the profiles. I'll get in touch with @rtyr when they are ready for wider use.

jaddel, 2022-10-21:
Is there currently any kind of workaroud to implement such thing for an IDEX printer?

3DSmitty, 2023-02-05:
FYI, hopefully this helps. This is what works for my Dual Extruder setup:

Under Filament Settings -> Custom G-code -> Start G-code:

; Filament Start gcode
M109 S[temperature] T[current_extruder] ; set temperature and wait for it to be reached

Under Filament Settings -> Custom G-code -> End G-code:

; Filament End gcode 
M104 S{temperature[0]+standby_temperature_delta} T[current_extruder] ; set temperature

Then Under Printer Settings -> Custom G-code -> Tool change G-code:

; Tool Change gcode
G1 X1.0 ; Move to safe location

So after slicing my final gcode looks like this:

; Filament End gcode 
M104 S185 T0 ; set temperature

; Tool Change gcode
G1 X1.0 ; Move to safe location
T1
G92 E0
; Filament Start gcode
M109 S235 T1 ; set temperature and wait for it to be reached

lukasmatena, 2023-02-06:
Please, try out [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3). The reworked logic of ooze prevention and the extra parameter `idle_temperature` in filament settings should make you all happy. Please, read [change log of 2.6.0-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha2) before trying it out (especially "Multi-material printing section". Feedback very much appreciated.

Also, #9235 is an exact duplicate of this issue.

Closing.

gbkwiatt, 2023-02-28:
I am changing from cura  to prusa. And I am trying to understand, why do I have to enable oozing prevention to  be able to use idle temp (even in 2.6.0-alpha) ? I want to be able to have Idle temp, when extruder is parked, but I don't really want the shield. Why those options are dependant on each other ?