prusa3d/PrusaSlicer #367: slic3r and slic3r-prusa fight over ~/.Slic3r/slic3r.ini format

hyperair, 2017-06-18:
### Version
- Slic3r git 98fca909
- Slic3r Prusa 1.35.5+dfsg-1

### Operating system type + version
Ubuntu 17.04

### Behavior
Switching between slic3r and slic3r-prusa causes the printer, filament and print profiles to reset to the first one in the dropdown. It seems slic3r-prusa appends ".ini" to the values in the `[presets]` section but slic3r does not.

bubnikv, 2017-06-19:
We did not change anything about the handling of the ini files. What do you recommend?

hyperair, 2017-06-19:
Drop the .ini suffix perhaps? Or maybe look into cherry-picking some commits from the main Slic3r.

bubnikv, 2017-11-02:
Fixed with https://github.com/prusa3d/Slic3r/commit/dd9e1aff7023dab9e33f4710bd808387f103e998
It will be part of the 1.38.xx release.

Also the whole presets, config bundles and application config were rewritten into C++.

bubnikv, 2017-11-02:
implemented, closing.

closed 2017-11-02 by bubnikv