prusa3d/PrusaSlicer #368: Compilation failure on debian

mavu, 2017-06-20:
### Version

version_1.35.5-3-g7ca02bda

### Operating system type + version
Linux debian stretch (testing)

### Behavior
perl BUILD.pl --sudo 

fails with 

src/libslic3r/Model.cpp: In member function ‘void Slic3r::ModelObject::split(Slic3r::ModelObjectPtrs*)’:
src/libslic3r/Model.cpp:644:20: error: ‘FLT_MAX’ was not declared in this scope
         float z0 = FLT_MAX;
                    ^~~~~~~
error building src/libslic3r/Model.o from 'src/libslic3r/Model.cpp' at /usr/share/perl/5.24/ExtUtils/CBuilder/Base.pm line 174.
FAIL
! Installing ./xs failed. See /home/mavu/.cpanm/work/1497960073.31322/build.log for details. Retry with --force to force install it.

Hugoz12, 2017-06-20:
Don't install with sudo. It will mess up a lot (at least it did for me, and I couldn't fix it yet on my raspbian). If it doesn't work without sudo you need to install some requirements. 
Can you post the content of the mentioned build.log?

What I would try but without any warrenty for correctness:
Install xs manually by changing to the xs folder and execute 'perl BUILD.pl'

mavu, 2017-06-20:
tried without sudo, same error, 

[build.txt](https://github.com/prusa3d/Slic3r/files/1089572/build.txt)

Hugoz12, 2017-06-20:
Try 'perl BUILD.pl --force'
If that fails too then install xs first as written above.

bubnikv, 2017-06-20:
There is a bug, which I will fix tomorrow. It compiled on my Windows box,
but neither gcc nor the OSX compiler do.

mavu, 2017-06-21:
Why do I always pick the one commit that can't compile on my box... :)

Thanks for the heads up.

bubnikv, 2017-06-21:
Fixed with https://github.com/prusa3d/Slic3r/commit/0bd2bb1e8e1a1b83ae1bd2ed5fafed6e7643e7e3

closed 2017-06-21 by bubnikv