prusa3d/PrusaSlicer #369: Modifier for support generation only available on object level

rheamagna, 2017-06-21:
### Version
1.35.5 PE

### Operating system type + version
W10 64bit

### Behavior
When selecting an object and adding a modifier to it (under object settings), it is not possible to modify the generation of support.
[image]

In the picture above, I want to generate supports under the cavity in the front, but I do not want to generate it for the recessions in the vertical pieces in the rear. By moving the reddish cube over the section that I don't want supports, I'd like to be able to give it the modifier "Do not generate support". At the moment, it is only possible to select this on the entire object.

So summarizing, it is possible to have multiple objects on a build plate, some with and some without support. It is not possible to use the modifier tools to disable (or enable) support generation on sections of an object.

iirc, this was a possibility in the past, not sure if it is a new feature.

bubnikv, 2017-06-21:
> iirc, this was a possibility in the past, not sure if it is a new feature.

I doubt it.

MoeweX, 2017-06-21:
Still, I think this feature could be used as a work around for #22 and might not require to much effort.

bubnikv, 2017-07-07:
I know Cura has something like this.

steelydev, 2017-07-17:
I would like to add the suggestion:

Allow any modifier mesh to either 'generate support material' or 'do not generate support material'.

hoegge, 2018-02-13:
This is really a feature request and is also wished for and described in the original Slic3r version [https://github.com/alexrj/Slic3r/issues/2570](url)

bubnikv, 2018-09-27:
> Allow any modifier mesh to either 'generate support material' or 'do not generate support material'.

Implemented in Slic3r 1.41.1-beta

closed 2018-09-27 by bubnikv