prusa3d/PrusaSlicer #370: Rotating about X axis causes first layer perimeter to be wrong

pigrew, 2017-06-22:
### Version
Prusa Slic3r 1.35.5-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
When rotating a model about the X axis by 90 degrees, the first-layer perimeter is no longer correct.

When the model was rotated in the CAD software (and not in Slic3r), the first layer is correct. Layers after the first layer are always correct.

[image]

Also, the support material pattern in the upper-right of the screenshot is different for the two instances. I'd expect them to be the same.

#### STL/Config (.ZIP) where problem occurs

[RotatePerimeterBug.zip](https://github.com/prusa3d/Slic3r/files/1095598/RotatePerimeterBug.zip)

bubnikv, 2017-06-26:
I am afraid this is not really a bug in slic3r, but an unusual effect of numerical rounding. You just have your object aligned too closely with the slicing plane of Slic3r, and the rotation causes slight perturbation of the vertices, pushing them randomly below or above the horizontal plane. If you had the horizontal plane a bit lower, you would not get this weird effect.

pigrew, 2017-06-26:
I looked at this a bit further, and it seems that it is an error in the STL file, and slic3r did report that the manifold had to be auto-repaired.  I'm starting to think that it's a bug in FreeCAD's STL export.

closed 2017-06-26 by pigrew