prusa3d/PrusaSlicer #374: Infill width not calculated correctly on stars pattern

bardiir, 2017-06-24:
### Version
1.34.1.24-prusa3d

### Operating system type + version
OSX 10.11.6 El Capitan

### Behavior
I've got this part and when I fill this with the stars pattern whcih I normally use at a layer height of 0.1 then it does calculate the infill as very thick extrusions basically overflowing the part when printing.
[image]

If I switch to for example 3D honeycomb then it does calculate correctly.
[image]

Also, if I switch to sparse infill every two layers then it does calculate correctly too.
[image]

#### STL/Config (.ZIP) where problem occurs
[Archiv.zip](https://github.com/prusa3d/Slic3r/files/1099951/Archiv.zip)

Hugoz12, 2017-06-24:
Probably related to #361
Try giving all extrusion widths values a definite width.

bardiir, 2017-06-24:
Yes that actually does take effect quite nicely of the issue. I guess the extrusion width should not be calculated over the layer height anyway. Wouldn't it make more sense to calculate it over the nozzle width?

bubnikv, 2017-07-19:
I have changed the "sane" extrusion widths to be in the ballpark of 1x to 1.125x of the nozzle diameter by default, independent of the layer height. 
https://github.com/prusa3d/Slic3r/commit/ac672d9dc592dc94ea1efdfeecdb75efc0a19233

The original code by @alexrj tried to maximize the print speed by maximizing the throughput of the plastics through the nozzle, which led to increasing extrusion widths for thinner layers. The approach to increase nozzle throughput may sound reasonably, but for the Prusa i3 MK2, we rather do the opposite: We increase the line width for thicker layers. Indeed, for thinner layers less material could be extruded due to a high viscosity of the plastic, and thicker lines need to be extruded into thick layers for the extrudates to fuse reliably. So as a reasonable compromise, the 1x to 1.125x of the nozzle diameter seems to be fine.

closed 2017-07-19 by bubnikv