prusa3d/PrusaSlicer #376: Feature request: Zoom primitives

m-Th, 2017-06-27:
### Version
Slic3r Prusa Edition
Version: 1.35.4-prusa3d-win64
Build: Slic3r-1.35.4-prusa3d-win64-full-201706151614
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread

* We do not have Zoom to All, Zoom to Plater and Keyboard shortcuts for these commands, including for Zoom In/Zoom Out.

Proposed Implementation:

Add in the View Menu the following menu entries with the following shortcuts

Zoom In__________ +
Zoom Out _______  -
Zoom to All ______ *
Zoom to Plater ___ /

Perhaps only "Zoom to All" needs explanation: it will zoom to enclose entirely the object(s) added to plater.

_Is this a new feature request?_

Yes

thess, 2018-01-06:
I too would like to see keyboard zoom as well. Not everyone uses a mouse or a multi-finger enabled touchpad. I have hacked in some menu driven zoom code but would REALLY like to see keyboard acceleration as well. VERY frustrating to use without this feature. :+1:

bubnikv, 2018-12-11:
We have implemented the following hot keys:
Zoom In__________ i
Zoom Out _______ o
Zoom to Plater ___ b
Zoom to Object ______ z

These hot keys will be part of the upcoming Slic3r 1.42 alpha release. Closing.

closed 2018-12-11 by bubnikv

m-Th, 2018-12-12:
@bubnikv ; Hmmm... I think that you're speaking about 1.43, isn't it?