prusa3d/PrusaSlicer #377: GUI Feature request: Add splitters (annoying behavior)

m-Th, 2017-06-27:
Slic3r Prusa Edition
Version: 1.35.4-prusa3d-win64
Build: Slic3r-1.35.4-prusa3d-win64-full-201706151614
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
The drop-down combo list with the names of profiles is too narrow. See below at the red arrow how the name is truncated.

[image]

Please make it wider or, ideally, add splitters (http://docs.wxwidgets.org/3.1/overview_splitterwindow.html ) at the green arrows in order to allow the user to make the list as wide as he wants. Of course the same thing should be implemented in all tabs: **Print Settings**, **Filament Settings** and **Printer Settings**

Of course, the position of the splitters should be persisted in an ini file or something similar.

m-Th, 2017-07-08:
...or ditch splitters and persistence engine entirely. Do something like this:

[image]

*Add*/*Delete* buttons together with the Combo Box to have their own pane at top.

In this way you can align them as you wish or simply give them the desired BIG width without messing with current profile values' pane.

fiveangle, 2017-07-11:
I agree this has always been a minor but annoying poor UX implementation

bubnikv, 2017-07-11:
Implemented with https://github.com/prusa3d/Slic3r/commit/a7153c67e6c548a7b923e9a59c717005c69fb551
by changing layout of the page.

closed 2017-07-11 by bubnikv