prusa3d/PrusaSlicer #379: Feature Request: Add to Plater: Ctrl+O

m-Th, 2017-06-27:
### Version
Version: 1.35.4-prusa3d-win64
Build: Slic3r-1.35.4-prusa3d-win64-full-201706151614
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
Annoying to endlessly click on Plater > Add...

Please add the Ctrl+O (Open) shortcut to this action.

ManuGithubSteam, 2017-08-06:
You can just drag and drop your files into the slicer window as well :-)

m-Th, 2017-08-06:
Of course. :) 
But for this one needs to have a big monitor or many monitors. FTR, I have 2 monitors of 27" and one of 40", so I am not _really_ in the market for this feature. :)

However because it is very efficient (IOW the ratio between effort of implementation and the result if very good) and because it help in streamlining the workflow (the Open dialog opens in the same directory and a keystroke is much quicker than mouse) and because is being discoverable (it appears in menus & tooltips) it determined me to propose this.

bubnikv, 2017-12-05:
Implemented with https://github.com/prusa3d/Slic3r/commit/2eeca93a971b07248caed9730dc65fb51bb7d76c

closed 2017-12-05 by bubnikv