prusa3d/PrusaSlicer #382: exponentional values generated as G1 F-1.95058e+006 causing problems

funbart, 2017-06-27:
### Version
Slic3r 1.34.1.3-prusa3d-win64

Hi, I got in some Gcodes extreme retraction. After some searching it turned out that there were Gcodes generated for F values as:

G1 X213.704 Y91.600 E79.91304
**G1 F-1.08173e+006**
G1 X213.663 Y93.643 E79.91247

Some wrote the feed value in exponential form, but there is no exponential form in gcode, the gcode interpreter will interpret it as G1 F-1.08173 E6. So in this case the E goes from 79mm to
6mm then back to 79mm.

Gcode, stl and profile can be downloaded [here](https://www.craftbot.nl/wp-content/uploads/2017/06/extreme-retraction.zip). 

Bart

funbart, 2017-06-29:
Am I wrong?
Anyway, what I do now is searching for e+ in the gcode and removing that line.....

bubnikv, 2017-07-05:
I cannot reproduce this issue with 1.35.5. Would you please retest?

This issue may have been fixed when I rewrote the G-code generator in the
1.35 series due to our multi-material release.

funbart, 2017-07-06:
Hi, tried it with 1.35.5 as well (win 64), Win 10 updated
same results: some G1 F-2.00573e+006 like values.

Can it be because of some regional settings?

bubnikv, 2017-07-07:
What language settings are you running on?

bubnikv, 2017-07-07:
Sorry, I still cannot reproduce your issue with Slic3r PE 1.35.5. Are you sure you retested with 1.35.5?

funbart, 2017-07-09:
I thought I was certain, but when retrying I didn't get the e+ error.
Sorry!

bubnikv, 2017-07-10:
Great, thanks. I will close this issue, please reopen if you encounter this issue again.

closed 2017-07-10 by bubnikv

gruvin, 2017-08-09:
Just encountered this myself, in Slic3r `v1.36.2-prusa3d`.

See attached zip file, containing the Slic3r-prusa settings used, the resulting gcode and the version of `3DBenchy.stl` used, which may have encountered MeshMixer and thus is possibly not pristine.

Line 1373 of `3DBenchy.gcode` and many more like it. Just search 'F-'.
```
G1 F-136105
```

The fault is 100% reproducible, after quitting and relaunching Slic3r, importing (drag drop) the STL file and exporting G-Code.

[Bad-GCODE-Slic3r-v1.36.2-prusa3d_macos.zip](https://github.com/prusa3d/Slic3r/files/1210302/Bad-GCODE-Slic3r-v1.36.2-prusa3d_macos.zip)

gruvin, 2017-08-09:
@funbart I've re-opened this as new issue #463. Problem seems to happen only with `Max volumetric speed` set to a non-zero figure -- possibly only if the print tries to exceed the figure present? Hope that helps.