prusa3d/PrusaSlicer #384: Firmware retraction

pcsentinel, 2017-06-28:
1.35.5- prusa

question, why does selecting firmware retraction disable the wipe before retract function? I can see no disadvantage to have the tool wipe while the firmware does the retract. apart from the fact that the slic3r doesn't know how much is being retracted, but that could be set within the wipe before retract parms.

Sebastianv650, 2017-06-28:
FW retract and wipe can't be executed together. During a wipe move, you want a retract and a horizontal (the wipe) move to be done. If you tell the FW to do a FW retract with G10, it will just retract the filament with no option to do a simultaneous wipe move.

bubnikv, 2017-07-05:
I agree with @Sebastianv650. During the wipe movement a partial retract is
performed.

Also why are you using the firmware retract? What firmware? I believe the
Slic3r PE 1.35 provides a more finer retract settings than what you can
achieve with your firmware.

Sebastianv650, 2017-07-05:
> Slic3r PE 1.35 provides a more finer retract settings than what you can achieve with your firmware.

@pcsentinel that's true. FW retract had an advantage for some time, because you was able to set a prime speed seperate from retract speed. But the current Slic3r alows you to do the same thing.
FW retract has some advantages for calibration tests, when you want to change the retract settings within a print to see what happens. But aside from this, I see no further advantage any more.

pcsentinel, 2017-07-16:
bubnikv, sorry yes your right just being dumb. Yes the software retract works great. 

Sebastianv650 agree about only using it for calibration now.

closed 2017-07-16 by pcsentinel