prusa3d/PrusaSlicer #385: Part registration for multiple extruders

sniglet, 2017-06-29:
### Version
1.33.8-prusa3d_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
MacOS Sierra 10.12.5

### Behavior
Adding additional files for additional extruders does not work as expected. It's not possible to add parts that maintain Z-offsets to permit stacking of parts for multimaterial. Offset tools don't appear to be active yet (not a problem), but 'Auto Center Parts' selector does not act as expected. 

There is no obvious way to import multiple files and maintain registration of parts. 

[Slic3r_config_bundle.ini.zip](https://github.com/prusa3d/Slic3r/files/1111895/Slic3r_config_bundle.ini.zip)

[image]
[image]

[Archive.zip](https://github.com/prusa3d/Slic3r/files/1111891/Archive.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-06-29:
Does this help?
http://manual.prusa3d.com/Guide/Gcode+preparation+for+Multi+Material/366

sniglet, 2017-06-29:
Not quite, the example has all base Z elevations identical. In the bench
model included in the bug report, the secondary extruder model doesn’t line
up when pulled onto the build plate.

You know how you can sink a part beneath the plate to continued a print?
This almost needs a negative Z offset to get the one file to line up with
the other...

There’s a GUI for file offset changes, but it doesn’t appear to be active
yet.

bubnikv, 2017-07-07:
First, the Slic3r PE 1.33.8 is very old. You should use the current one, which is 1.35.5.

I have tested with the multi-material 3D Benchy. It works, if I load the boat bottom part first, then I add the other parts through the object dialog. If I start with a part that is not aligned with the bed itself (for example the chimney), then the other parts are not aligned as they would collide with the print bed.

So Slic3r shall likely be modified to lift the window once the boat bottom is loaded. Is it the only problem you are having, or do you have some other issues?

sniglet, 2017-07-07:
I’m away from home on vacation, when I left, I was able to multi material
print ABS, but was awaiting replacement PTFE hot end parts to finish
troubleshooting issues with PLA...I’m having some success with the MM
upgrade, and am willing to be extremely patient while things develop.

This is not a trivial upgrade and has lots of changes to the toolchain,
kinematics, material tolerances, etc.

bubnikv, 2018-02-12:
No response, closing.

closed 2018-02-12 by bubnikv