prusa3d/PrusaSlicer #387: Blobs on the Heatbed while doing Heatbed Mesh Leveling

m-Th, 2017-06-30:
### Version
Slic3r Prusa Edition
Version: 1.35.5-prusa3d-win64
Build: Slic3r-1.35.5-prusa3d-win64-full-201706151818
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
See here for description of the problem and fix: - the fix is changing the GCode generation in Slic3r:
https://github.com/prusa3d/Prusa-Firmware/issues/121

That's why I posted it here.

m-Th, 2017-07-05:
It seems that the fix has some problems. I continued the thread over there.

bubnikv, 2017-07-05:
The problem with your proposal is, we would likely have to introduce another temperature settings for each filament, as the "soft enough, but not leaking" temperature will be certainly filament specific. And there will be risks, that if the temperature is too low and the blob too hard, the Z switch will not trigger.

m-Th, 2017-07-05:
Yes - I experienced it practically. 
...but in any case it is a serious problem which must be fixed. Now I am hunting with a spatula to remove the blobs left in the mesh leveling points (especially in center) because in most cases till the job reaches there these blobs become hard and are dangerous for the quality of the print.

Two ideas:

1. The "soft enough, but not leaking" cannot be expressed in a relative way for example by a macro like [heatbed_temp - 20] or as an editing field in Slic3r' Filament Settings | Filament | Temperature | Extruder? (ok, the profiles must be updated)

2. After mesh leveling phase park the head at maxX, maxY, Z:10cm (for ex.) and let the LCD to display a message **"Click to actually start the print"**. Besides allowing the user to clean the heatbed, this will serve also as a confirmation before starting the job (you know, everyone sees that he has forgot something after job starts - cleaning the heatbed, changing the filament/settings etc.)

bubnikv, 2017-07-05:
> 1) The "soft enough, but not leaking" cannot be expressed in a relative way for example by a macro like [heatbed_temp - 20] or as an editing field in Slic3r' Filament Settings | Filament | Temperature | Extruder? (ok, the profiles must be updated)

There is no math expression evaluation implemented in Slic3r macro processor. Also a generic -20 will likely not work in general.

> After mesh leveling phase park the head at maxX, maxY, Z:10cm (for ex.) and let the LCD to display a message "Click to actually start the print". Besides allowing the user to clean the heatbed, this [...]

I believe you can program that in the start G-code using the standard Marlin features, namely the M0 code.

http://marlinfw.org/meta/gcode/

m-Th, 2017-07-05:
> I believe you can program that in the start G-code using the standard Marlin features, namely the M0 code.

Thanks! Good to know! I'll have to do some tests now...

bubnikv, 2017-07-11:
Are you fine with the solution?

m-Th, 2017-07-12:
Hummm... http://marlinfw.org/docs/gcode/M000-M001.html says

M0 [P<ms>] [S<sec>] [string] 

But *M0 S300 Click to continue* doesn't show anything on the LCD...

Which is the separator for strings? Double quotes? Single quote? 

A collateral discussion here is: **What happens in the case of a syntax error?**

The best would be a **Preflight feature** (you know, like Compiler Warnings) to check for such problems, check if the print is inside of the margins of the heatbed, different objects are colliding between them, objects in the air (because in the STL there are other objects lower than them, minimum thickness walls, perhaps a check for overhangs (run a simulation with the Generate Supports engine etc,) etc.

bubnikv, 2017-10-03:
Our Marlin firmware fork will support M0 with both time and a string in the upcoming release.

We have a feature request for integrating a math expression parser into Slic3r, which is my preferred way of handling these complex feature requests.
https://github.com/prusa3d/Slic3r/issues/438

I am closing this issue, and I will reference this issue at the other one.

closed 2017-10-03 by bubnikv

bubnikv, 2017-11-19:
Please discuss your proposals on the conditional G-code syntax at https://github.com/prusa3d/Slic3r/issues/438