prusa3d/PrusaSlicer #388: Unicode char like for example "ł" crush app when used in profile name

porlock, 2017-06-30:
### Version
Slic3r-1.35.5-prusa3d-win64-full-201706151818

### Operating system type + version
Windows 10

### Behavior
Unicode char like for example "ł" crush app when used in profile name.
Problem does not occurs on older Slic3r-1.34.1.24-prusa3d-win64-full-201705201440 version.

_Is this a new feature request?_
No:

bubnikv, 2017-07-07:
How does Slic3r crash?

I tested on Windows 10 switched to Czech language, saved a printer profile with Czech specific characters, then reopened slic3r and all was well.

It is a known issue though, that Slic3r will not handle non-ASCII7 chraracters if they do not fit the current code page. That means, you should have the Polish language selected system wide before trying to use Polish characters in file names in Slic3r.

porlock, 2017-07-07:
Well i notice that it does not happens always, but if You try hared enough You can do the "trick" i manage to do this when saving printer settings (before i save successfully other profiles with "ł" sign) look at the screen.

bubnikv, 2017-07-11:
Does it help, if you edit lib\Slic3r.pm
find the decode_path subroutine and remove the following line:
       unless utf8::is_utf8($path)
(you have to remove the line, but keep the trailing semicolon).

The missing utf8 library seems to be the issue, but also this place seems to be the only one where it is called, so I don't want to bundle the library with Slic3r.

bubnikv, 2017-07-11:
Fixed with https://github.com/prusa3d/Slic3r/commit/6f28818f872e14c511a7ccdf73d5d6b5d53189aa
by avoiding the triggering of the UTF8 Perl library, using the explicit upper / lower case Perl regexes instead of the /i suffix.

closed 2017-07-11 by bubnikv