prusa3d/PrusaSlicer #389: [Feature Request]Separate value for Z-axis motion speed

Bubolz, 2017-06-30:
### Version
Slic3r 1.35.5 - Prusa Edition

### Operating system type + version
Win7

### Behavior
It would be interesting if there were the possibility of setting the Z-axis speed separately for Machines capable of achieving high X Y displacement speeds like CoreXY.

Sebastianv650, 2017-06-30:
Maybe I'm not understanding you correctly, but the FW is defining the maximum speed limits. As printers use XY plane for printing, why should the slicer influence the Z speed different from the travel speed?

Bubolz, 2017-06-30:
My machine has a cubic mount, with kinematic style Makerbor replicator, but with bowden in the car. It moves bed (Z-axis) with a TR8 screw driver

The X and Y axes with belt and bowden can reach about 500mm / s but the Z axis (screw driver) can not go beyond 120mm / s. With equal speed for all axes I have to limit the movements of X and Y to 120mm / s. (If I set higher speeds my Z loses steps)  I can resolve using M commands but it would be better to have an additional configuration parameter.

Note: The Simplify 3D has separate speed settings for XY and Z.

[image]

Sebastianv650, 2017-06-30:
The maximum axis speed as you need it is something that has to be controlled within the FW. Check and set you FW configuration accordingly. All Simplify can do is also call the M code (M203 Z..), which means overwriting the FW setting which wouldn't be necessary if it has been properly configured before.
Maybe it will also use it for z-only moves, but for all combined moves there is no way in gcode to set a dedicated z speed.

Or if you want a slicer solution anyway, you can add the M203 code to your printers start gcode section. That will do exactly the same as Simplify does.

Bubolz, 2017-06-30:
Thanks Sebastian! Regarding the M203 command is what I'm currently doing, but I will make a speed correction in firmware.

bubnikv, 2017-07-05:
Thanks @Sebastianv650 for taking care of this issue. I believe we can close it.

closed 2017-07-05 by bubnikv