prusa3d/PrusaSlicer #391: New feature request: Shutdown after finishing the print job

m-Th, 2017-07-01:
### Version
Slic3r Prusa Edition
Version: 1.35.5-prusa3d-win64
Build: Slic3r-1.35.5-prusa3d-win64-full-201706151818
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
When we have a long run job which finishes somewhere in the middle of the night, the machine (the electronics etc.) remain open. The users would like an option to have the machine automatically shut down at the end of the job (many times is better to have an electronics & firmware reset and cooldown).

### How to implement
* Add a check box in Printer Settings | General which would say "Shutdown at the end of the Job" (or similar)
* if checked, the program will simply send at the end of the job a M0 G-Code.

m-Th, 2017-07-05:
> if checked, the program will simply append at the end of the job a M0 G-Code.

Correction: is actually M081 - http://marlinfw.org/docs/gcode/M081.html

bubnikv, 2017-07-05:
Why don't you just add this G-code to the end of your end G-code section?

I am a bit opposed to add everyone's favorite feature as it would make the application overly complex and for the user difficult to understand.

bubnikv, 2017-07-07:
Sorry, closing for now. There is too much work and you can implement your feature using the end G-code.

closed 2017-07-07 by bubnikv