prusa3d/PrusaSlicer #392: Crash after rotating sliced part in layer view

Hugoz12, 2017-07-02:
### Version
1.35.5-prusa3d-linux64

### Operating system type + version
Linux 3.16.0-4-amd64 #1 SMP Debian 3.16.43-2 (2017-04-30) x86_64 GNU/Linux

### Behavior

- Add a part
- view it in "Layers" tab
- slice it
- rotate it by clicking the 45degree buttons
--> Slic3r crashes
(When the GUI is started via terminal it becomes unresponsive without an error message in the terminal. Ctrl+C is needed to close it.)

* _Expected Results_
- layers disappear until you slice again
This actually happens when you first slice the part then change to "Layers" tab and rotate. Reappears after slicing. But crashes if you rotate after that.

Auto-processing deactivated. Tested with 2 models.
If you can't reproduce I'll provide the parts and config.

bubnikv, 2017-07-05:
- Add a part
   - view it in "Layers" tab
   - slice it
   - Did you switch to the layer view here? Or the 3D preview? The "Layers"
   tab has the buttons disabled on Windows. Maybe they shall be disabled on
   Linux as well?
   - rotate it by clicking the 45degree buttons
   --> Slic3r crashes
   Still in the layer view? Or the 3D preview?
   (When the GUI is started via terminal it becomes unresponsive without an
   error message in the terminal. Ctrl+C is needed to close it.)

   - *Expected Results*

   - layers disappear until you slice again
   This actually happens when you first slice the part then change to
   "Layers" tab and rotate. Reappears after slicing. But crashes if you rotate
   after that.
   If you slice, then go to layer view, then back to 3D view, delete an
   object, load another one and then repeat the "crashing" scenario once
   again, will it crash again?

Hugoz12, 2017-07-08:
> - Add a part - view it in "Layers" tab - slice it - 

> Did you switch to the layer view here? Or the 3D preview? The "Layers" tab has the buttons disabled on Windows. Maybe they shall be disabled on Linux as well?

No I switched to the layers tab before slicing.

> The "Layers" tab has the buttons disabled on Windows. Maybe they shall be disabled on Linux as well?

Good idea. Maybe it's just an oversight on Linux because it doesn't show anything anyway.

> - rotate it by clicking the 45degree buttons --> Slic3r crashes 

> Still in the layer view? Or the 3D  preview? 

Yes all done in Layers tab.

>  If you slice, then go to layer view, then back to 3D view, delete an object, load another one and then repeat the "crashing" scenario once again, will it crash again?

Yes it still crashes.

bubnikv, 2018-03-15:
@FidelCapo was not able to reproduce this issue. He tested on 1.35.5 and on our latest integration build 1.40.0-alpha.249.

There is an issue with the cancellation of the slicing core, which is still partially written in Perl. I suppose this issue will go away in 1.41, when the complete slicing backend will be rewritten into C++.

bubnikv, 2018-12-11:
We hope we fixed your issue by refactoring the background processing code.
Please retest with 1.42 alpha once it is released this or the next week. Closing.

closed 2018-12-11 by bubnikv