prusa3d/PrusaSlicer #393: Dual extrusion independant extrusion width

pcsentinel, 2017-07-02:
### Version
1.35.5
Feature request

The system allows for different nozzle sizes for each extruder, however there is only one set of extrusion width settings. It would be useful to be able to set extrusion width by extruder.

bubnikv, 2017-07-05:
I agree your request makes sense, but it is quite a lot of work and I wonder how many users would actually use this feature.

I wonder what use case would require this feature?

You could set the extrusion widths per volume of an object, so you can set a different extrusion widths for each extruder already, though I agree it is cumbersome.

pcsentinel, 2017-07-16:
Hi, I use it for dual extrusion with a soluble support material, to speed up printing I use a 0.4 nozzle, but I use a 0.25 nozzle for the print so that I can get finer detail. I currently do set it to 0.4 in the advanced settings, but it means I cant use your auto extrusion width calculations.

Chaos99, 2017-09-27:
I second that request. Even when printing with the same material for both extruders, I'd like to use one for infill, where the 'combine infill every x layers' can be cranked up higher when using a wider nozzle (same for solid infill, top/bottom infill) as opposed to perimeters, where I use a small nozzle for details.

gnydick, 2019-06-08:
+1 for me. This is a common pattern, large nozzle for infill and supports and small nozzle for detail.

masterjuggler, 2021-06-09:
This looks like it hasn't been touched in a long time, so I wanted to say +1 from me.

I have a dual extruder printer and would like to print with multiple materials at once more easily. For example soluble supports or PLA and TPU with a mechanical connection. Right now I have to manually change every setting individually for every part.

bubnikv, 2022-01-20:
Does anyone use nozzles of differing diameters regularly? While I understand the potential theoretical benefit, how does it look like in reality? Did you try it just once and then never tried again, or do you use it regularly, or do you just want to try it?

pcsentinel, 2022-01-20:
I would like to use it on a regular basis. 0.4mm for the soluble support material, as this is not easy to get through a 0.25mm nozzle and 0.25mm for detail work.

Kachidoki2807, 2022-01-20:
> Did you try it just once and then never tried again, or do you use it regularly, or do you just want to try it?

I wish I could, but I can't since my favorite slicer is PS, and I don't have time nor the desire to tinker a new printer profile for another slicer which might support this feature. This does not worth it.

My thoughts about this topic here https://github.com/prusa3d/PrusaSlicer/issues/5589#issuecomment-1017810977

masterjuggler, 2022-01-20:
Actually I do use two nozzle sizes, on a very frequent basis, and have been since I last posted in June. I usually have a 0.4mm nozzle on one extruder and a 1.4mm on the second. My job involves a lot of prototyping and printing of machine parts that are relatively time sensitive, so it's extremely handy to have both already on the printer. I'd like to be able to print a bed full of parts at once, some with a small nozzle for details, some with a large nozzle for strength and speed

I also sometimes have the same diameter on both extruders and print multiple materials on the same part as I already mentioned. In fact, the use case of combining PLA and TPU with a mechanical connection has come up more than once, and they require VERY different print settings.

Being able to set a totally different set of print settings per nozzle or per part would make things a lot easier. I waste a lot of time changing settings on a part by part basis instead of just selecting a profile currently.

bubnikv, 2022-01-20:
@masterjuggler
Did you ever use the two different nozzles on the same object?

masterjuggler, 2022-01-20:
If by object you mean a single manifold mesh, then no and I don't believe that's what is being discussed, unless I've misunderstood the point of this github issue.

If you mean what enters prusa as a multi part object and comes off the printer as a single physical object, then I have many times used each extruder for a different material, and would like to easily set different extrusions widths as necessary. 

For example, I recently printed an auger to move powder that used PETG for the core and TPU as a wiper on the edges, fixed to each other with a dovetail along the flights. It would have made my life much simpler if I could have just set a different print setting profile, extrusion width and all, to each part of the object.

I've come to understand that I am talking about something more general than the original request, which is strictly about extrusion width, and bringing whole profiles into the equation. At the very least, I believe this would be a good implementation of the request, or would at least make it far easier to utilize a multi-extruder setup.

Kachidoki2807, 2022-01-21:
> Being able to set a totally different set of print settings per nozzle or per part would make things a lot easier. I waste a lot of time changing settings on a part by part basis instead of just selecting a profile currently.

I like this idea but it certainly won't be so easy to handle the overlapping into the settings. I have the same annoying problem but with the filament profile too. I mean, I often use twice the same filament profile, but need a slightly different settings for one of them depending on the print type, like temp / cooling adjustment. Currently it is not possible to do that, unless to save at least one profile into a temporary user profile. It is not possible to have the same profile in "default" state on an extruder, and in "modified" on another extruder nor both in "modified" state. The problem is getting worse as the extruder count increase.

pcsentinel, 2022-01-21:
Yes, well one size for support material and one for the actual model

Regards

Paul Clarke

bubnikv, 2022-01-21:
> @masterjuggler

I also sometimes have the same diameter on both extruders and print multiple materials on the same part as I already mentioned. In fact, the use case of combining PLA and TPU with a mechanical connection has come up more than once, and they require VERY different print settings.

Which print parameters do you have to change? I suppose it is mostly the print speed and possibly bridging speed and bridging flow, right? The extrusion widths stay the same for PLA and TPU, right?

We are actively looking into a way to combine materials and nozzles, however to make everybody happy will likely make the profile management a great pain. I urge anybody to look into what Cura guys did. Their system is versatile, however extremely complex. Do we need the same level of versatility?

bubnikv, 2022-01-21:
@masterjuggler 

> If you mean what enters prusa as a multi part object and comes off the printer as a single physical object, then I have many times used each extruder for a different material, and would like to [...]

Would you want to use different nozzle diameters on the same object?

Why do you want the PLA to be extruded with the different extrusion widths from TPU? What is your motivation? Properties of the two resins or the traits of the actual part to be printed?

masterjuggler, 2022-01-21:
> Which print parameters do you have to change? I suppose it is mostly the print speed and possibly bridging speed and bridging flow, right? The extrusion widths stay the same for PLA and TPU, right?

For multi material prints I change most infill settings like density and pattern (gyroid gives TPU a very isotropic squish), I'd very much like to be able to set max volumetric speed per part, perimeter overlap, elephant foot compensation, and probably a few others I can't think of off the top of my head.

Generally for multi material prints I use the same nozzle diameter in each extruder, yes. There are times when it is beneficial not to though, mostly in the pursuit of faster prints when time matters.

> We are actively looking into a way to combine materials and nozzles, however to make everybody happy will likely make the profile management a great pain. I urge anybody to look into what Cura guys did. Their system is versatile, however extremely complex. Do we need the same level of versatility?

It's been several years since I last gave Cura a really good look, I'll have to take some time later to do so. There are definitely features that PrusaSlicer has though that I've used a LOT since introduction, like the paint on supports and seams. I may have to start using multiple slicers in my workflow lol, though I'd really much rather not have to resort to that.

masterjuggler, 2022-01-24:
> Why do you want the PLA to be extruded with the different extrusion widths from TPU? What is your motivation? Properties of the two resins or the traits of the actual part to be printed?

Just saw this post. Going back to my example of the auger, I needed to use a very narrow extrusion width for the TPU wiper to make it as flexible as possible so it wouldn't jam inside the tube, but I wanted to use a much larger nozzle for the PETG core so it would print in a fraction of the time it took.

neophyl, 2022-01-25:
I don't have dual extrusion at the moment.  However when I get my XL one of my primary use cases is larger practical items, with those I would like to be able to print the perimeters (or possibly only the outer perimeter) with the smaller nozzle for detail and use a larger nozzle for the infill (and possibly the normal perimeters).  Mostly will be the same material in each extruder though for me.
With that larger print area anything we can do to reduce print time is going to be a factor.

Googliola, 2023-05-24:
+1
I use 0.6 for TPU (less clogging etc) and 0.4 for PVA supports on the same object. 
I also use 0.6 and 0.4 for infill / perimeter with various materials.

With the advent of Prusa XL., this is a almost a must-have feature. What if I wanted to use a drill, pick-and-place or some other non-filament tool on one of the five heads? 

I posted a rough idea on how I would PrusaSlicer expect to handle such a scenario here https://forum.prusa3d.com/forum/postid/657717/

Would appreciate if there was a timeline as I ordered a PrusaXL with 5 tools, but without proper software support, I dont see much sense to own such a device at all and rather stick to my IDEX.

hickengr, 2024-06-06:
I would like to print faceplates of enclosures with text.  i have done text adn faceplate with a 0.4mm nozzle, the I have to make the print large enough to make it readable.  If I can use a 0.25mm nozzle, I could print more clear in smaller sizes while printing enclosures with either 0.6mm or 0.4mm nozzles.

closed 2026-09-29 by prusaslicer-bot[bot]