prusa3d/PrusaSlicer #394: Ozze Shield & tool offset

pcsentinel, 2017-07-02:
1.35.5

The ooze shield feature seems to take account of tool offsets for moving both heads outside the shield if they are set in the extruder settings. However if your tool offsets are set in the firmware then they are not taken into account. It would be useful to have a feature where the offsets are only applied to the moves to place the heads outside of the shield or to have a specific park location that can be set.

bubnikv, 2017-07-05:
What firmware are you using?
Cannot you just use the extruder offsets in Slic3r?
Isn't it possible to switch the extruder before printing the ooze shield?

I am a bit opposed to add a feature for a single user. Also the ooze shield in Slic3r is not quite what it should look like, it should follow the contour of the object.

pcsentinel, 2017-07-16:
I have adjusted the offsets by a firmware call and it solves the issue, but it just feels wrong in principle. The firmware offsets are often used for other things outside of slicing, including auto bed levelling etc.

gregspruce, 2017-08-09:
I would love to see some more customization options regarding the ooze shield and cool down/warm up for tool changes. It is borderline necessary to be able to ensure my tool change doesn't leave one nozzle hanging *inside* the skirt, potentially even oozing directly onto a perimeter.

I am against doing things in gcode when it should be saved in firmware - nozzle offsets won't change, why set them everytime and then have to worry about setting them for every slicer I use or every profile per slicer.

closed 2026-09-29 by prusaslicer-bot[bot]