prusa3d/PrusaSlicer #395: [Enhancement] Per Filament G-Code

sarf2k4, 2017-07-02:
As per my previous feature request to "per filament g-code" to be implemented, I just thought that this actually might give too much of a "filament profiles" especially if a user owns/handles multiple 3d printers.

As of now, the idea is that it is tied to per filament but if I were to shift the filament to another printer, it is either I create another "filament profile" or change the g-code according to the printer.

Example
```
Filament X on Printer A, the z offset is at -1.4mm, While Filament Y on Printer A, z-offset is at -0.8mm

Filament X on Printer B, the z offset is at -2.4mm, While Filament Y on Printer B, z-offset is at 1.5mm.
```
In the above example, There should be at least 2 "filament profile" for each Filament X and Filament Y just to be used on both printers; A and B. What if the user has about 5 or 10 printers? Note that both Filament X and Y above are different material

I would like to add a bit of an enhancement to the "per filament g-code" as follows

The slic3r will create a unique "printer ID" and will get displayed on the printer tab>general section.
In the "Filament g-code", the slic3r will then read a new tag/variable inside there then will only add that specific multiple lines under the new variable which is the "Printer ID" identifier. Example as follows

Printer A ID is 3
Printer B ID is 4

Enhanced Filament G-code:
```
[ID:3] ;Refers to Printer A
set z-offset -1.4mm
set K value to 0

[ID:4] ;Refers to Printer B
set z-offset -2.4mm
set K value to 40
```
So based on the g-code example above, the slic3r will read the matching ID to the selected printer profile. If there are no "Printer ID" issued but there are g-codes, it will be applied to any g-code that are not specified under that specific ID.

This reduces the amount of "Filament Profile" created just to cater for multiple printer's using same material but has their own specific g-codes.

What do you guys think?

fiveangle, 2017-07-02:
I had this same revelation yesterday when moving to test the same print using LIN_ADVANCE on my MK2 (PLA K=50)  on my Printrbot Metal (PLA K=70) when my excitement sank.

The only way I can see this working is to make it feature-specific (not generic GCODE area) so that one can have a filament profile, and a printer profile.  

For example, you've calibrated each filament to your MK2 standard, then you buy a MK2-MM and find all the K constant calibrations require +50.  One would need a specific setting for both.

Example:
-MK2 Printer Profile K = 0
-PLA Filament Profile = 50
=> M900 K50 in GCODE

-MK2 MM Printer Profile K = 50
-PLA Filament Profile =50
=> M900 K100 in GCODE

_Why's everthang gots ta be so compulated ?_  😉

sarf2k4, 2017-07-02:
I was only focusing on the z-offset for each filament. Other than that is also the lin_advance feature. I don't know more than that yet. 

I take it that from your proposal above, is just to add a text box on both printer profile and filament profile for the K value to be used on Lin_advance.

To be frank I haven't used the filament g-code yet, not even once. This is mainly because the current g-code to adjust z-offset were set to absolute instead of subtract/addition on the addition

Sebastianv650, 2017-07-02:
The basic problems whe are seeing here (and also in #393) is that the given settings groups Print, Filament and Printer is not realy fitting the needs, especialy when it comes to multi color / multi nozzle or features like LIN_ADVANCE and per-filament / printer z offsets.

The proposal above proves this with a work around. A real solution would be to think about the settings in the existing groups. Are they realy only depending on the printer - filament - print quality settings?
In many cases you have to answer the question with no, only to give some examples:

Printer:
- Z-Offset: Depends on filament material (used print temperatures to be precise)
- Retraction settings: Depends on filament up to some point. Think about PLA vs. Ninjaflex. While 0.5mm will work for PLA, Ninjaflex will not even recognise this small retract I bet.

Filament:
- Cooling settings. While this belongs to the filament on a first view, it might not in reality. Fot example I have a highly effective dual-side cooling on my TAZ. The Prusa i3 will melt my prints with the low fan settings I'm using on the TAZ. The cooling power needed is filament based, but we are setting a fan speed here, not knowing the cooling power of the printer.

Print:
- Line widths depends on the nozzle diameter, therefore it depends on the printer.
- Speeds: Can depend on the filament type (again think about PLA vs Ninjaflex) and the printer itself. For example if one has a high quality V-Slot rail system, the other cheap one might use loose bearings or weak steppers that can't go with high speeds at the wanted quality level.

I guess an implementation to solve all this dependencies would confuse 99% of the average users. You would need to create a complete table of dependencies. Now we could set static values per profile (the ones that realy only depends on print, filament, printer) and other values which are variable due to combinations of other settings. To given an example based on the filament cooling settings this could mean:
- The values in the filament tab changes when you select another printer. If you have not entered a value for the specific combination, Slic3r has to notify you and force you to enter something. For beginners, that would mean to never know which settings are used when and why. Or
- There are more tabs, showing specific values for the given combination. At least, this would clearly state that this values are not static.

In any case, this will increase the complexity a lot. On the other side, creating now work arounds for any possible situation will end the same way.

I hope somebody has more useful ideas ;-)

sarf2k4, 2017-07-02:
Heheh... i really hope there are better solutions to this, although not
that critical but can solve dilemma of 3d printer owner of variable
z-offset changed by filament type

What if, these enhancement for z-offset will be put inside printer section,
and then another sub section that has a series of textbox for the z-offset
according to per filament type. User dont have to manually put filament
gcode like what i proposed in the first post, instead insert these textbox
with numbers, then in the start gcode will have something like this `m851
z[z_offset]` where the [z_offset] will be taken from the new z-offset
section. Then these values will be chosen based on which filament type that
will be used. Since in filament section, there are some setting about the
material selection/identifier has been introduced. To my preference, i
would like to have a base/default z-offset value, then slic3r would just
add/subtract based on the other variables. For example,

```
Z-offset in relative mode: boolean type (false is absolute value will be
used)
Base z-offset: -0.700
Pla z-offset: 0.200
Abs z-offset: 0
```
In the example above the setting were in relative mode, and me for example
uses abs as main material so i took the z-offset value from the printer
itself and put into the base z-offset. If i were to change filament type to
pla, then the base will add the pla offset value, resulting in -0.500 for
the offset issued by slic3r in the [z-offset].

Of course, lin_advance could follow this method as well.

Bit of enhancement to dave's proposal =)

fiveangle, 2017-07-03:
Agree with @Sebastianv650 that what we're propsing here are work-arounds that do not resolve the issue at the root, but they are simple and effective for the main current pain points for the user.  A handful of easy-to-implement additive/subtractive fields shown on the "Advanced" interface only is a lot easier that re-architecture of entire preferences model.  Unless @bubnikv has a better idea :)

bubnikv, 2017-07-07:
The problem is, there will always be a user or a situation, for whom the parameter hierarchy does not fit. There is no way around it, the parameters are simply not strictly hierarchical. The only correct solution is to have a set of parameters per printer+filament+print. This is what we do with PrusaControl.

@alexrj implemented an interesting way to override parameters from one group by another. You may check his builds and then give proposals based on the experience with his implementation.

I may implement scripting (math expressions, conditions) in the G-code sections. With the free-form scripting, you could implement whatever you want.

bubnikv, 2017-10-03:
> The only correct solution is to have a set of parameters per printer+filament+print. This is what we do with PrusaControl.

Cura does it as well, at least for their machines. Their setting hierarchy is so complex and undocumented, that I doubt that any other manufacturer is able to replicate what Ultimaker did for their machines.

bubnikv, 2017-11-19:
Please discuss conditional g-codes at https://github.com/prusa3d/Slic3r/issues/438

bubnikv, 2017-11-28:
@sarf2k4 We have basically implemented your request with a new G-code conditional syntax. We have applied the new conditional G-code syntax to the linear advance parameter, you may check the preliminary "merged" settings, where all our printer profiles (MK2, MK2MM, MK2MM-single, MK3) are merged together into a single tree. We also integrated the concept of "compatible print profiles" and "compatible filament profiles" with the printer profile, as implemented in the original slic3r.

Here are the merged presets.
https://github.com/prusa3d/slic3r-settings/tree/merged

This is the conditional syntax applied to the linear advance:

M900 K{if printer_preset=="Original Prusa i3 MK2 MultiMaterial"}200{elsif printer_preset=="Original Prusa i3 MK2 MM Single Mode"}200{else}30{endif}

The syntax is rudimentary and it will certainly improve. I have implemented a bare minimum, as there is a risk, that I may come up with some quirky syntax in a hurry, which people will start using, and I would not be able to go back. I am open to the conditional language proposals. Most likely, there will be a Slic3r 1.33.8 alpha release tomorrow.

bubnikv, 2017-11-28:
@Sebastianv650 

> The basic problems whe are seeing here (and also in #393) is that the given settings groups Print, Filament and Printer is not realy fitting the needs, especialy when it comes to multi color / multi nozzle or features like LIN_ADVANCE and per-filament / printer z offsets.

We may discuss the parameters one by one, whether they will fit the print, filament or a printer profile, but the settings dependencies will never be perfectly one-directional, and these exceptions (the reverse dependencies) will need to be covered by some tables of exceptions. A reasonable solution is the conditional G-code syntax above, another band aid are the compatible print & filament profiles (this is a less elegant solution if it requires duplication of many profiles).

In the future, the "system" presets (the presets fine tuned by Prusa Research) will be described by a hierarchy of JSON or XML files in a similar way to what Cura does. There will be no UI for mingling with these JSON or XML hierarchical configs.

If there are no objections, I would like to close this issue.

sarf2k4, 2017-11-29:
If the setting were to be based on json and xml like cura does, in my
opinion that thelic3r will be very limited, for example cura can do dual
material printong but only if ultimaker machine were to be selected, in the
past i briefly searched how to use cura with dual material printing on
other machine the ultimachines, it is close to impossible.

If json and xml will be implemented, i guess im okay with it just that not
to be too strict. If editing those files are possible, i suggest add some
menu "edit json/xml" or "open json/xml/profiles folder" rather having to
browse the folders manually.

bubnikv, 2017-11-29:
> If the setting were to be based on json and xml like cura does, in my
opinion that thelic3r will be very limited, for example cura can do dual
material printong but only if ultimaker machine were to be selected, in the
past i briefly searched how to use cura with dual material printing on
other machine the ultimachines, it is close to impossible.

The specification is not finalized yet. My idea for the Slic3r read-only JSON or XML hierarchical configs is to generate a read only ".ini" style of a config behind the scenes, and to offer the user to store his modifications into the usual .ini files. So you would have both ways to configure your printer and you could continue to use the .ini files only.

I believe the Cura profiles are open, but not documented. It is the documentation that is missing, not the functionality.

bubnikv, 2017-12-05:
We have a conditional G-code syntax, which implements the initial request. I propose to close this issue. Any objections?

If you want anything of the above discussed to be implemented, I propose that you open a specific issue for each request. Thank you.

bubnikv, 2018-02-12:
Many of the configuration issues discussed in this topic could be covered by the new conditional G-code to some extent. Closing this issue.

Please open new issues for specific topics, which are difficult to cover with the conditional G-code syntax.

closed 2018-02-12 by bubnikv