prusa3d/PrusaSlicer #396: Possible fix for stringy material on the MK2 multi-material upgrade

hathawsh, 2017-07-03:
I have PLA (from Makergeeks) that extrudes well at 230-235C. (At lower temperatures, it often jams.) It works with the multi-material upgrade, except that when unloading filament, it often makes a stringy tip inside the extruder, leading to a jam when the printer tries to reload the filament later. (Curiously, the PLA is *not* stringy when extruded normally.)

I discovered today that a simple modification of the gcode for unloading the filament seems to resolve the issue and avoid jams. Then current unload procedure first pulls the filament 15mm at maximum speed. I manually edited that part of my gcode to pull out 15mm, dwell for 500 milliseconds, push 15mm back in, then pull 15mm out again. This has the effect of compressing the string inside the extruder into a small ball so the rest of the unload procedure works as designed and forms a good tip.

In other words, I basically did a search and replace. I changed all instances of: 

```
G1 E-15.0000 F5000
```

to:

```
G1 E-15.0000 F5000
G4 P500
G1 E15.0000 F5000
G1 E-15.0000 F5000
```

It seems like this string compression method might be appropriate for all users of the MM upgrade. It might even make PETG compatible with the MM upgrade; I plan to test that hypothesis this week.

hathawsh, 2017-07-03:
Update: my testing shows this idea definitely helps Makergeeks PLA and PETG, but it hurts TPU and possibly normal PLA. Will test more.

bubnikv, 2017-07-05:
I forwarded your proposal to Alex, who developed the filament switch sequences. The current unload procedure rams the molten plastic out of the melt zone before unload, which is the same procedure as the one used for the filament switch.

bubnikv, 2017-07-05:
I discussed quickly with Alex. According to him, your proposal has been tried, but it did not work reliably. 
Please note that the current filament switching procedure is a result of many months of testing and parameter tuning.

closed 2017-07-05 by bubnikv

hathawsh, 2017-07-05:
Thanks for discussing it. Unfortunately, the testing apparently did not include the kind of filament I'm using and the current filament switching procedure reliably jams (after only 1-2 filament changes) when using most of my filament spools. Fortunately, my solution seems to solve the jamming and I can write a little program that edits the gcode. I'll test more.