prusa3d/PrusaSlicer #398: [Feature Request] Save variable layer height profile

hollandm, 2017-07-03:
### Version
1.33.8-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
I frequently use the variable layer height functionality both with the layer editing button and with [right click part -> part settings -> layer options]. Most of the the things I print use the same variable layer height settings but every time I have to slice a part I have to re-enter the information. I would love to have a way to save layer height settings as part of a profile or at minimum be able to easily copy/paste them into the dialog.

_Is this a new feature request?_
Yes

#### STL/Config (.ZIP) where problem occurs
N/A

bubnikv, 2017-07-05:
The layer height is being stored into the AMF file. Just export the platter
to the AMF file.

bubnikv, 2017-07-07:
Already implemented as part of the AMF format. Closing. If you have objections or you found a bug in the current implementation, feel free to reopen this issue.

closed 2017-07-07 by bubnikv