prusa3d/PrusaSlicer #399: MIssing possibilty to do 2 perimeter vase mode...

dembach, 2017-07-03:
### Version
1.35.5
### Operating system Mac OS 10.12.5

### Behavior
* i want to do 2 perimeters as vase mode to get a water-proof vase on problematic designs
* vase mode disables the possibility to do 2 perimeters
 
* _Expected Results: 2 perimeters in vase mode
* _Actual Results : only 1 perimeter

_Is this a new feature request? yes

Sebastianv650, 2017-07-03:
That's not possible. Think about how you would lay down two or more perimeters without starting and stopping the continous extrusion.

fiveangle, 2017-07-03:
It is technically possible to do but would require setting a seam position, and retracing back to the last "layer" origin point begin laying down the adjacent "outer" layer.

But unsure what this gains since the primary driver behind vasemode is a seemless end-piece.

There are other optimizations that allow for water-proof vase (increase extrusion multiplier, slow print down, increase print temperature, decrease layer height, all perturbations of these values :)

fiveangle, 2017-07-03:
now that i'm thinking about this, while it would not be possible to perform a 2-extrusion-width vase mode, it would be possible to do a 3-extrusion-width one, by travelling 1/2-loop up the "vase" for the first point off the base layer, then loop back to the origin, then turnback again to continue a full loop... at end of full loop, turn back 1/2 loop to meet end of first 1/2-loop, then turn back and begin a full loop again.  This would produce no actual Z scar (although there would surely be visibility of the two 1/2-loop ends meeting) but I imagine the result could be quite interesting, visually, since these ends should match up, or spiral up through the print.

if I were technically inclined enough, I think it would be worth tackling... "sturdy vase mode" :)

fiveangle, 2017-07-03:
```
>_______  _____>____   _____>_____  _____>______  _____>__...
____<___||__<__ __<__||__<__ __<__||__<__ __<__||__<__
|_______>______||_____>_____||_____>_____||____>______|
```

dembach, 2017-07-03:
al@fiveAngle, yeah, quite interesting the 3 perimeter version..... 
it is not always nice to do more extrusion width , because you might loose details. so the attempt with 3-extrusions is a very good and nice ways to get a water-resistant printout without overextrusion.
thanks for looking at this feature request 
best regards
Markus from germany

Sebastianv650, 2017-07-04:
When you have to go back in some way, this means you will not have a continous increasing z height but az layer shift as in normal mode. In this case you will have something like a visible seam,that's why i said it's not possible.
The advantage of the vase mode is a perfect visual appearance, which you will loose of you do it as you described it.

fiveangle, 2017-07-04:
I think you misread what was stated:

> This would produce no actual Z scar (although there would surely be visibility of the two 1/2-loop ends meeting) but I imagine the result could be quite interesting, visually, since these ends should match up, spiraling up through the print.

Sebastianv650, 2017-07-04:
Maybe I can't imagine what you want to describe :)
But I remeber an issue in original Slic3r or PR Slic3r where something like this was already discussed. There was also some helpfuls sketches over there. I will try to find it again, the end result was that it wouldn't work. But again, I might be wrong..

fiveangle, 2017-07-04:
To help you visualize, it's only the first loop off the flat adherence base that would require special treatment, and would have to be printed not with an ever-increasing Z-height as the nozzle moves linearly around the loop, but would need to be printed with the layer height increasing as if a radial swept around the "vase" were constantly increasing Z as it cut through the model perimeter (meaning the nozzle would move up while moving in one radial direction, and decrease for segments where it was moving the opposite, only for the first loop off the flat base). The end result would be a continuously-increasing layer height cut throughout the entire perimeter.  Once the math of the first loop off the base were set, every loop afterward would be constant extrusion height (as in normal vase mode).

You don't need to worry about this with std vase mode because you are never backtracking, but with this modification it is absolutely possible (from a pure physics perspective — I've of course not seen what the results would look like but it's not hard to imagine a positive outcome :)

fiveangle, 2017-07-04:
Come to think of it, normal vase mode would work as well and you wouldn't need any special treatment for the first loop off-base, but the intra-perimeter layer heights would be different ,  which may negatively affect the visual quality (or could improve, I guess we won't know until someone tries it,  which could certainly be done with standard slicing software if the model was generated  with imperceptibly thin cuts through the walls,  similar to how the linked model above is  —  if I have time I may attempt to generate a cylinder demonstrating this behavior using openscad as it should be  relatively easy.  I'm curious how it would look :)

Sebastianv650, 2017-07-04:
When you try it - also do demonstrators with inward and outward sloped walls. Most "avoid seem" ideas fail in this cases.

fiveangle, 2017-07-04:
The model would be limited to designs that can print with "print outside perimeters first", but otherwise I don't envision other limitations.  I'll keep you posted.

mavu, 2017-07-06:
How about making the wall thicker by doing a tight zig-zag instead?

fiveangle, 2017-07-06:
that would be soooooooooo sloooooooooow and i doubt would be visually appealing (but I guess you never know untill you try.  it certainly would be able to print some crazy overhang designs :)

bubnikv, 2017-07-07:
I am inclined to close this discussion. There are 150 open issues on Slic3r PE and I doubt, that a zig-zag perimeter will be visually appealing. The purpose of a vase mode is to remove the seam, but the only proposals given in this discussion actually added new scars.

fiveangle, 2017-07-07:
++

bubnikv, 2017-10-03:
It was an interesting discussion, but there are more pressing issues to be fixed.
Closing.

closed 2017-10-03 by bubnikv