prusa3d/PrusaSlicer #400: Not lifting extruder after printing complete individual objects

mdjurfeldt, 2017-07-04:
### Version
e4a7131

### Operating system type + version
Debian stretch

### Behavior
* When the option "Complete individual objects" is checked and the print is finished, the extruder is not lifted above the tallest individual object but instead crashes into objects when moving the print bed towards user.

* Set radius to 60 mm and height to 25 mm (for Prusa MK2) and check "Complete individual objects". Add a tall and a low object and press "Arrange". The lower object should be at the front of the bed. If it is not, add the objects in the opposite order. Print.

* At end of print, I expect the Z-axis to lift the extruder at least above the taller object.

* When print is finished and the Y-axis moves towards the user, the extruder is not lifted above the tall object but crashes diagonally into it.

#### STL/Config (.ZIP) where problem occurs
[screwnut.zip](https://github.com/prusa3d/Slic3r/files/1122733/screwnut.zip)

bubnikv, 2017-07-05:
The "complete objects" shall be printed in the order from the lowest to the
tallest object, so the print head should not crash into the already printed
objects when homing at the end of the print. Does it work differently in
the recent Slic3r PE builds?

mdjurfeldt, 2017-07-05:
From my perspective as a user, it does seem to work differently. In my
example with a crew (taller) and a nut (lower), the order is determined by
the order in which I add them to the plater.

bubnikv, 2017-07-05:
There was a typo in the G-code generator after I rewrote it from Perl to C++.
The following check-in fixes the order of the objects, so they are printed in the order from the smallest to the tallest object.
https://github.com/prusa3d/Slic3r/commit/e78839d2fcbf716b1d68969e99ef7bb0e2d37073

closed 2017-07-05 by bubnikv