prusa3d/PrusaSlicer #402: New feature: Double & Triple density external perimeters

frank-hw, 2017-07-05:
### New Feature Request
Smooth variable layers are nice, but the feature I wish for most is Double & Triple density external perimeters.

### Behavior
For each layer, external perimeters print first with 2 or 3 layers at 1/2 or 1/3 of the layer height.
Internal perimeters and infill print (2 or 3 times faster) at the regular layer height.

cerdeira, 2017-07-05:
Take a look at "Combine infill every n layers"

bubnikv, 2017-07-05:
There is a "combine infill" feature in Slic3r, though I personally never tested it.
The idea of printing the internal perimeters with twice the layer height is not new, but I wonder how well the double layer height extrusion would squish where it we would like it to be squished. I am personally quite skeptical.

BTW it looks like holiday season started, the discussions are getting more active.

cerdeira, 2017-07-05:
I use combine infill with infill layer height 0.25mm and perimeters 0.12mm, and it is ok.
The solid layers are 0.12 for both.
After printing the solid layers it then prints 1 perimeter at 0.12, 1 infill at 0.25 and then another perimeter at 0.12 again.

The part exterior is very nice and the interior is just the usual.
But it gave me some work to tune the printer for layer height 0.12mm, high print speeds can be a problem.

rennne, 2019-02-25:
This could be very nice if you want to print strong parts with maybe 6 perimeters.
Adding perimeter gives a lot of strenght: [https://www.youtube.com/watch?v=AmEaNAwFSfI](https://www.youtube.com/watch?v=AmEaNAwFSfI)

keeping the outermost(1) and maybe the next perimeter(2) at 0.12 and 3,4,5,6 at 0.25 would cut print-time substantial but still keep a lot of detail on the outside

JohnnyDeer, 2019-10-11:
Hi, this issue has already been solved. In latest versions of PrusaSlicer is possible to set density of perimeters.
I will close this issue in framework of improving the clarity of the issues list. Thank you for understanding.

rennne, 2019-10-11:
I believe this should not be closed

This feature request is not fulfilled by that function, because it changes all perimeters on that layer, not just the one on the outside

JohnnyDeer, 2019-10-15:
There is possible to change density just for external perimeters 
[image]

wavexx, 2019-10-15:
I don't get it. Could you explain again?

rennne, 2019-11-02:
If you view it the other way around it would be called:
"Combine Inner Perimeter for every 2 or 3 Layer"

similar to "Combine Infill every 2 or 3 Layer"

this results in shorter printing time, while the amount of visible outer perimeters stays the same

futureconstruct, 2022-06-06:
> If you view it the other way around it would be called: "Combine Inner Perimeter for every 2 or 3 Layer"
> 
> similar to "Combine Infill every 2 or 3 Layer"
> 
> this results in shorter printing time, while the amount of visible outer perimeters stays the same

Great explanation!

blurayne, 2023-11-03:
Also got the problem of weak walls when trying to print with 100% infill. model was a compartment for my cupboard door to hold some tools. The walls either didn't stick together or material extrusion was too low so I got holes in the wall (100% infill !!!!)

I experimented a bit with internal and external extrusion width, but walls didn't stick together. I had a bit more success when pumping general extrusion to 300%.

The problem I have: while I can print a box at 100% perfection like the printer does for spiral vase -  i am disappointed by what the expected outcome of a model with 100% infill is. Even 2mm walls just don't stick together with default settings and I can throw the print into garbage.

CobraA1, 2024-02-04:
There is another vote for this option [in Reddit](https://www.reddit.com/r/prusa3d/comments/1aimwif/slicing_with_outer_layer_being_a_fraction_of_main/). Actually a nice idea, since only the outside perimeter is the most visible and inner perimeters are usually hidden.

Aquilux, 2026-09-24:
this is still a useful feature, especially now with the release of INDX and the usefulness of printing with radically different sizes of nozzle

Jan-Soustruznik, 2026-10-01:
Hello, thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about this feature. Thank you, and have a nice day.