prusa3d/PrusaSlicer #405: Separate Copies

m-Th, 2017-07-08:
### Version
Version: 1.35.5-prusa3d-win64
Build: Slic3r-1.35.5-prusa3d-win64-full-201706151818
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
Steps:
1. Put some small objects on platter and (if applicable) split it in parts.
2. Go to Object | Increase Copies... to fill the plater with objects
3. Select a copy and move it somewhere on the Plater
4. Go to Object | Rotate 45 Clockwise...

Expected:
Only the said copy should be rotated or - MUCH BETTER - provide us the possibility to **Separate Copies Apart** in order to allow us to have two entirely separate objects.

Actual:
All the copies are rotated 

For example, in the screenshot above, I **cannot** rotate the two copies of the digit "1" separately.

[image]

_Is this a new feature request?_
Yes.

bubnikv, 2017-07-12:
Would you propose a UI for that?

m-Th, 2017-07-12:
Yes. Put it in two places:

- **Logical**: Add a menu entry in menu *Object* called *Separate Copies* after *Set number of copies...* and assign Ctrl+\ as a shortcut

- **Convenient**: Add a menu entry in the current object's right-click menu in the same place. See here:

[image]

m-Th, 2018-12-12:
@bubnikv : Do you mean 1.43, right?

EXOgreen, 2018-12-25:
Suggest closing this issue as the alpha "fixes" this @bubnikv

m-Th, 2018-12-26:
@EXOgreen , @bubnikv "Separate" isn't present in 1.42a , just the rotation

bubnikv, 2019-03-05:
Splitting of instances is implemented in Slic3r 1.42.0-alpha7. Closing.

closed 2019-03-05 by bubnikv